rtlabs-com/p-net #1: Sample App GSDML

svenhaitjema, 2019-10-07:
Dear Hans-Erik,

Would it be possible to upload a sample GSDML file that matches the code in the sample_app directory? 

Kind regards,

Sven-Erik

nakarlsson, 2019-10-09:
I'm working on it

svenhaitjema, 2019-10-09:
Thank you very much! I can setup a test with TIA 15 and S7-1200 PLC. Please let me know if you need any feedback from that test :)

jstavats, 2020-01-10:
Hi,

Any chance you could give a quick description of what is needed to be done to use this sample?  I've updated the IP address and the ethernet device (eth0) to match my machine, but the PLC is giving a "Module configured but not available (inserted and configured module types are not the same) - Expected type: IO Device (12345) - Current Type: Not Available".   The sample looks to be setting the iodevice to 12345.

Thanks,
Jay

svenhaitjema, 2020-01-13:
Dear Jstavats, 

There is a known issue with the library + GSDML file when used with Siemens TIA. Could you specify what your setup is? ( So PLC brand/type, and software studio version? )

Please see the conversation in this pull request, maybe it relates to the error you are observing.

Kind regards,

jstavats, 2020-01-13:
Hi, 

My current test setup is older, the PLC is Siemens CPU315-2 PN/DP and Software is SIMATIC STEP 7 Professional 2010 v5.5 +SP3. 

Jay

commit 8e278c5a (2019-10-09): Add sample_app IO Device GSDML file

Simple sample IO Device GSDML file
Three plugable modules
8 Bit Inputs
8 Bit Ouputs
8 Bit Inputs/8 Bit Outputs

fix #1
closed 2020-01-29 by hefloryd