rtlabs-com/p-net #2: Compilation fails on linux due to compiler warning

svenhaitjema, 2019-10-07:
Dear Hans-Erik,

The source code fails to build on Linux (raspbian) due to a compiler warning: 

The 'fix' for now is removing: -Werror from line 34 in: cmake/Linux.cmake
so: set(CMAKE_C_FLAGS "-Wall -Wextra -Wno-unused-parameter -Werror") 
becomes: set(CMAKE_C_FLAGS "-Wall -Wextra -Wno-unused-parameter") 

The compiler warning is caused by line 41-45 of src/osal/linux/osal_udp.c:

local = (struct sockaddr_in) {
         .sin_family = AF_INET,
         .sin_addr.s_addr = htonl(addr),
         .sin_port = htons(port),
      };

The sin struct appears to be missing a field initializer. 

Kind regards,

Sven-Erik

hefloryd, 2019-10-08:
We should initialise the missing field. Can you post the complete error message as we don't see it when building on Ubuntu or on Travis?

Alternatively we can do a memset on the whole struct prior to setting it.

svenhaitjema, 2019-10-08:
Thank you for your response, the full error is: 

[ 31%] Building C object src/CMakeFiles/profinet.dir/osal/linux/osal_udp.o
/home/pi/p-net/src/osal/linux/osal_udp.c: In function ‘os_udp_open’:
/home/pi/p-net/src/osal/linux/osal_udp.c:45:7: error: missing initializer for field ‘sin_zero’ of ‘struct sockaddr_in’ [-Werror=missing-field-initializers]
       };
       ^
In file included from /home/pi/p-net/src/osal/linux/osal_sys.h:27:0,
                 from /home/pi/p-net/src/osal.h:29,
                 from /home/pi/p-net/src/pf_includes.h:30,
                 from /home/pi/p-net/src/osal/linux/osal_udp.c:17:
/usr/include/netinet/in.h:246:19: note: ‘sin_zero’ declared here
     unsigned char sin_zero[sizeof (struct sockaddr) -
                   ^~~~~~~~
/home/pi/p-net/src/osal/linux/osal_udp.c: In function ‘os_udp_sendto’:
/home/pi/p-net/src/osal/linux/osal_udp.c:74:4: error: missing initializer for field ‘sin_zero’ of ‘struct sockaddr_in’ [-Werror=missing-field-initializers]
    };
    ^
In file included from /home/pi/p-net/src/osal/linux/osal_sys.h:27:0,
                 from /home/pi/p-net/src/osal.h:29,
                 from /home/pi/p-net/src/pf_includes.h:30,
                 from /home/pi/p-net/src/osal/linux/osal_udp.c:17:
/usr/include/netinet/in.h:246:19: note: ‘sin_zero’ declared here
     unsigned char sin_zero[sizeof (struct sockaddr) -
                   ^~~~~~~~
cc1: all warnings being treated as errors
src/CMakeFiles/profinet.dir/build.make:400: recipe for target 'src/CMakeFiles/profinet.dir/osal/linux/osal_udp.o' failed
make[2]: *** [src/CMakeFiles/profinet.dir/osal/linux/osal_udp.o] Error 1
CMakeFiles/Makefile2:981: recipe for target 'src/CMakeFiles/profinet.dir/all' failed
make[1]: *** [src/CMakeFiles/profinet.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2

hefloryd, 2019-10-08:
@svenhaitjema , can you confirm that this PR fixes your build?

PR #3 (2019-10-08): Initialise sin_zero field
Initialise the sin_zero field to avoid compiler warnings on some
architectures. Also, not setting sin_zero may cause undefined
behaviour on some architectures.

Fix #2.

Change-Id: I72b5667c59f480ab948a50695e34792dadb8e84f
commit 2a7b2ca7 (2019-10-08): Initialise sin_zero field

Initialise the sin_zero field to avoid compiler warnings on some
architectures. Also, not setting sin_zero may cause undefined
behaviour on some architectures.

Fix #2.

Change-Id: I72b5667c59f480ab948a50695e34792dadb8e84f
closed 2019-10-08 by svenhaitjema