rtlabs-com/p-net #5: Windows build failed

jstavats, 2019-11-26:
Hi,

I'm trying to build this under windows with VS2015 and Cmake3.15.5. When cmake gets into the sample_app directory it gives the following error and stops:

-- Configuring done
CMake Error at sample_app/CMakeLists.txt:14 (add_executable):
  No SOURCES given to target: pn_dev

CMake Generate step failed.  Build files cannot be regenerated correctly.

Is there something else I need to set?  It looks like that file is looking for ${APP_SOURCES} on that line which I can't seem to locate anywhere else.

Thanks,
Jay

hefloryd, 2019-11-28:
Yes, unfortunately the windows support is incomplete. Apart from the build system failing the windows os abstraction layer is currently incomplete. We are looking at unifying the os abstracfion layer for our stacks and should be able to fix this as part of that work.

jstavats, 2019-11-28:
It looks like the abstraction layer is most of the way there, it looks to only be lacking ethernet/udp calls, perhaps the linux layer can be ported/adapted for this?  What other support is incomplete?  Any estimate what kind of performance could be achieved when this is supported under windows?

pyhys, 2020-02-14:
We have now fully removed support for Windows unfortunately, as it would be very difficult to reach the timing requirements. See also #14

closed 2020-02-14 by hefloryd