rtlabs-com/p-net #14: Error in communication

svenhaitjema, 2020-02-07:
Dear All, 

Thank you for fixing the code related to the connection issues. The PLC&TIA14 is now able to connect to the p-net software, however I have the following observation making me believe something is still off: 

The PLC status alternates between 'connected' and 'connection error'. 

The error reporting on the p-net side is: 
[ERROR] PPM(479): iocs_len is zero
[ERROR] PPM(479): iocs_len is zero
[ERROR] PPM(479): iocs_len is zero
ABORT err_cls 253   err_code 5

OR 

[ERROR] PPM(479): iocs_len is zero
[ERROR] PPM(479): iocs_len is zero
[ERROR] PPM(479): iocs_len is zero
ABORT err_cls 0   err_code 6

Now I have also implemented the os abstraction code to read out a button( GPIO) and enabled it through the #IF 1 define. Now when I press the button i see that the pnet_input_set_data is called by I do not see the change reflected in the Tag value in the PLC. 

Kind regards,

Sven-Erik Haitjema

pyhys, 2020-02-13:
Normally the p-net stack on Linux sends a cyclic data frame approx every 3 ms. When this increases to around 7 ms (due to load), the Simatic PLC sends an alarm indicating that the timing requirements not are met. Then the cyclic communication is shut down for a few seconds, until a new start-up happens.

We are looking into using the PREEMPT_RT patches to improve the Linux real time properties.

pyhys, 2020-03-04:
You need to tweak the Linux real time properties to fulfil the Profinet timing requirements. We have started to document this on https://rt-labs.com/docs/p-net/linuxtiming.html

closed 2020-03-04 by hefloryd