rtlabs-com/p-net #22: Input or Output only slots

yannickasselin, 2020-02-18:
Hello,
Using the sample app, I was able to succesfully test Profinet communication with Codesys as a Profinet Master. I was also able to use different slots for communication by modifying the GSDML file and the main_linux.c file

Here's what I get in Codesys. "8 bits IN 8 bits Out" in slot 3 working fine:
[image]

But when I try to add "8 bits IN" in slot 1, I get this error when the PLC connects to the Profinet Stack:  "[ERROR] CMDEV(1604): api 0 exp slot 1 subslot 1 and dir 2 not found"

Here's what I changed in main_linux.c:
[image]
[image]

And here's what I changed in GSDML file:
[image]
[image]

I must be missing something?

Same problem if I try to add "8 bits OUT" in slot 2. I get this error:
 "[ERROR] CMDEV(1604): api 0 exp slot 2 subslot 1 and dir 1 not found"

What am I doing wrong?

pyhys, 2020-02-19:
We have managed to reproduce, and will look into it.

pyhys, 2020-03-16:
See pull request #36

PR #36 (2020-03-17): Fix bug when using output modules
Fixes #22
closed 2020-03-17 by hefloryd