rtlabs-com/p-net #26: Input Output Max Size

yannickasselin, 2020-02-19:
Hello,
After doing some tests with only 1 slot, I noticed that the max size I can use is 230 bytes for inputs and 244 bytes for outputs. I read on the internet that the max size for profinet is 1440 bytes total and 244 IN/OUT per slot. So why 230 bytes for inputs in this case?

Also I noticed that when using more than 1 slot then I have to divide the 230 and 244 bytes by the number of slots I am using. Is there a way to increase that size limit?

I already modified the GSDML file like this but it did not help. There must be something else?
<IOConfigData MaxInputLength="1440" MaxOutputLength="1440"/>

Thank you.

pyhys, 2020-03-16:
See pull request #35

pyhys, 2020-03-16:
Thanks for your bug report. Now fixed. Please try again.

closed 2020-03-16 by hefloryd