rtlabs-com/p-net #27: Why do I need "RecordDataList" for a slot to work?

yannickasselin, 2020-02-20:
Hello,
When using the sample_app with the supplied GSDML file, I can see the connection working fine with Codesys:
[image]

But if I comment out the "RecordDataList" part in the GSDML file like this, communication does not work anymore.
[image]
[image]
And I get this error:
ABORT err_cls 253  err_code 18

What else do I need to do to make it work without the "RecordDataList"?

I am trying to have multiple slots. A couple of "Input Only" slots and a couple of "Output Only" slots.
I think this should be a simple task but I am struggling.

I figured how to increase the PNET_MAX_MODULES in order to have more slots but now I need to figure out how to get more bytes IN/OUT per slots and get rid of the "RecordDataList".

Thank you

pyhys, 2020-03-17:
Hi, and thanks for your feedback.
After pull request #36 this should be functioning.

pyhys, 2020-03-24:
Hi, i can now confirm that it is functioning. I removed the RecordDataList part from GSDML-V2.35-rt-labs-IODevice-20200316.xml and tried it with the current version of the Linux sample app.

closed 2020-03-24 by hefloryd