rtlabs-com/p-net #34: Error when using many parameters

Svenson12, 2020-03-11:
When parameters are declared for sub-modules, the set values are transferred from the PLC to the device during the connection request using write requests. When the size of all parameters to be transferred is bigger than one Ethernet frame, this write request is fragmented.
Receiving these fragmented write requests seems to work, but if the required response becomes bigger than one Ethernet frame, the stack fails. The reason seems to be, that no measure is implemented in pf_cmrpc_dce_packet to handle this situation. I assume the correct reaction would be to send the response fragmented as well, but this is not implemented.
I tried to limit the size by setting "MaxSupportedRecordSize" in the GSDML file to a small value (e.g. 1024), but this violates the GSDML-specification and the GSDML file will be rejected when I try to load it in my S7 project. (I'm using an S7 CPU 1212C with TIA V16.)

Svenson12, 2020-03-21:
Can you confirm this issue?
Can you give an estimation when an update may be available?
Many thanks!

pyhys, 2020-04-02:
Yes, you are completely right, seel also issue #42 
This is a high priority for us.

Svenson12, 2020-04-14:
Thanks for your feedback. Can you already make a prediction when a new version could be available?

Svenson12, 2020-04-24:
Hello, I create a fix for this issue and wnat to shar it with you for review:
The function _pf_cmrpc_dce_packet_ must like like this:
`static int pf_cmrpc_dce_packet(
   pnet_t                  *net,
   uint32_t                ip_addr,
   uint16_t                port,
   uint8_t                 *p_req,
   uint32_t                req_len,
   uint8_t                 *p_res_final,
   uint16_t                *p_res_len_final,
   bool                    *p_is_release)
{
   int                     ret = -1;
   pf_rpc_header_t         rpc_req;
   pf_rpc_header_t         rpc_res;
   uint16_t                start_pos = 0;
   uint16_t                length_of_body_pos = 0;
   uint16_t                req_pos = 0;
   uint16_t                res_pos = 0;
   uint16_t                res_len;
   pf_get_info_t           get_info;
   pf_session_info_t       *p_sess = NULL;
   bool                    is_new_session = false;

   get_info.result = PF_PARSE_OK;
   get_info.p_buf = p_req;
   get_info.len = req_len;

   /* This function also sets get_info.is_big_endian */
   pf_get_dce_rpc_header(&get_info, &req_pos, &rpc_req);

   /* Find the session or allocate a new one */
   pf_session_locate_by_uuid(net, &rpc_req.activity_uuid, &p_sess);
   if (p_sess == NULL)
   {
      (void)pf_session_allocate(net, &p_sess);
      is_new_session = true;
   }

   if (p_sess == NULL)
   {
      /* Unavailable */
      LOG_ERROR(PF_RPC_LOG, "CMRPC(%d): Out of session resources.\n", __LINE__);
   }
   else
   {
      p_sess->get_info = get_info;
      memset(&p_sess->rpc_result, 0, sizeof(p_sess->rpc_result));

      /* Handle incoming RPC fragments: */
      if (rpc_req.packet_type == PF_RPC_PT_FRAG_ACK)
      {
         /* Re-route the parser to use the session buffer */
         req_pos = 0;
         p_sess->get_info.p_buf = p_sess->buffer;
         p_sess->get_info.len = p_sess->buf_len;
      }
      else
      if (rpc_req.flags.fragment == false)
      {
         /* This is the normal path where the incoming request is contained entirely in one frame */
         p_sess->buf_len = 0;
         p_sess->ip_addr = ip_addr;
         p_sess->port = port;  /* Source port on incoming message */

         p_sess->from_me = false;
         p_sess->activity_uuid = rpc_req.activity_uuid;

         p_sess->is_big_endian = p_sess->get_info.is_big_endian;
         p_sess->fragment_nbr = 0;
[... 237 lines cut ...]
            rpc_res.flags.fragment = false;
            rpc_res.flags.no_fack = false;
            rpc_res.flags.maybe = false;
            rpc_res.flags.idempotent = true;
            rpc_res.flags.broadcast = false;
            rpc_res.flags2.cancel_pending = false;
            rpc_res.length_of_body = 0;
            rpc_res.is_big_endian = p_sess->is_big_endian;

            pf_put_dce_rpc_header(&rpc_res, *p_res_len_final, p_res_final, &res_pos, &length_of_body_pos);
         }
         else
         {
            /* Waiting for the next fragment. */
         }

         ret = 0;
      }
   }

   /* Size of result data */
   *p_res_len_final = res_pos;

   return ret;
} `

Svenson12, 2020-04-24:
Can you confirm this fix?

pyhys, 2020-04-24:
Hi, great!
I will forward it to my colleague working on this part of the stack.

pyhys, 2020-05-05:
We have implemented a fix for this, see #62 
Please review.

PR #62 (2020-05-14): Implement fragmented DCE RPC responses
This allows larger responses, which is required when using more slots.

Closes #34
Closes #42
commit 9f948663 (2020-05-14): Implement fragmented DCE RPC responses

This allows larger responses, which is required when using more slots.

Closes #34
Closes #42
closed 2020-05-14 by hefloryd