rtlabs-com/p-net #44: how to evaluate this p-net io stack and what is the environment it requires

Rayapati1203, 2020-04-06:
First i am working on this  this library . I would like to know what is the environment i need to setup. 
My environment -
Host OS -is windows 10 
Guest OS - ubuntu18.04 on oracle VB guest OS. I am using ubuntu as my device IO.

Codesys IDE with Rasberry pi runtime is on windows10 and ubuntu as my device io.
As per the documentation, Rasberry PI hardware is required as IO Controller. But i did not have this hardware.

1. P-net library :  able to build and run it. but i found that the button files are not created for some reason.
-----------------
cfeuser@cfeuser-VirtualBox:~/pnet/profinet/p-net/build$ sudo ./pn_dev -v
[sudo] password for cfeuser: 

** Starting Profinet demo application **
Verbosity level:    1
Ethernet interface: enp0s3
MAC address:        08:00:27:6E:99:77
Station name:       rt-labs-dev
LED file:           
Button1 file:       
Button2 file:       
IP address:         10.0.2.15
Netmask:            255.255.255.0
Gateway:            10.0.2.1

Waiting for connect request from IO-controller
------------------
Can any please suggest me how to verify this without having rasberry PI hardware.

Thanks
Raj

I am facing the following issues1.As mentioned in documentation, i downloaded Codesys 3.5 and Rasberry pi runtime package. As i did not have rasberry pi hardware as IO controller.  In this case can you suggest me an alternative to evaluate this library if possible
2. P-net stack : Able to build and run it . 
But the the button files are not created in gpio folder.

Can any one helps to resolve my setup issue.

pyhys, 2020-04-08:
Hi Raj,
and thanks for your interest in the p-net Profinet stack.
In order to run the sample application (which is an IO-device) you need to run it together with an IO-controller (a PLC). One way to do it is to use Codesys on a Raspberry Pi as a PLC, but it is also possible to use some other Profinet cabable PLC as the IO-controller. For example Siemens PLCs can be used, and we describe it here: https://rt-labs.com/docs/p-net/use_with_siematic.html

The button files are typically existing Linux GPIO files, but you can also use plain files. But then you need to create the files manually in advance.

Rayapati1203, 2020-04-08:
Thanks for getting back to me and spend time to my request.

I have setup the environment yesterday - codesys(trial version) ,rasberry pi 3b+ as IO controller, P-net stack with sample app running on ubuntu18.04 linux VB in laptop.

I have created gpio files below as i did not find them in /sys.class/gpio in linux

- "class/gpio/gpio17/value"
- "class/gpio/gpio22/value"
- "class/gpio/gpio27/value"

[pnet_demo_log_druing_running.txt](https://github.com/rtlabs-com/p-net/files/4451561/pnet_demo_log_druing_running.txt)

i set it up and followed the step by step mentioned for codesys environment , i saw the callback message on p-net io device.
but i got the following error messages.

```
cfeuser@cfeuser-VirtualBox:~/pnet/profinet/p-net/build$ sudo ./pn_dev -v -l class/gpio/gpio17/value -b class/gpio/gpio22/value -d class/gpio/gpio27/value
[sudo] password for cfeuser: 

** Starting Profinet demo application **
Number of slots:     5 (incl slot for DAP module)
P-net log level:     3 (DEBUG=0, ERROR=3)
App verbosity level: 1
Ethernet interface:  enp0s3
Station name:        rt-labs-dev
LED file:            class/gpio/gpio17/value
Button1 file:        class/gpio/gpio22/value
Button2 file:        class/gpio/gpio27/value
MAC address:        08:00:27:16:8b:d0
IP address:         192.168.254.40
Netmask:            255.255.255.0
Gateway:            192.168.254.1

Waiting for connect request from IO-controller
Module plug call-back
  Pull old module.    API: 0 Slot: 0x0    Slot was empty.
  Plug module.        API: 0 Slot: 0x0 Module ID: 0x1 Index in supported modules: 0
Submodule plug call-back.
  Pull old submodule. API: 0 Slot: 0x0                   Subslot: 0x1      Subslot was empty.
  Plug submodule.     API: 0 Slot: 0x0 Module ID: 0x1    Subslot: 0x1 Submodule ID: 0x1 Index in supported submodules: 0 Dir: 0 In: 0 Out: 0 bytes
Submodule plug call-back.
  Pull old submodule. API: 0 Slot: 0x0                   Subslot: 0x8000      Subslot was empty.
  Plug submodule.     API: 0 Slot: 0x0 Module ID: 0x1    Subslot: 0x8000 Submodule ID: 0x8000 Index in supported submodules: 1 Dir: 0 In: 0 Out: 0 bytes
Submodule plug call-back.
  Pull old submodule. API: 0 Slot: 0x0                   Subslot: 0x8001      Subslot was empty.
  Plug submodule.     API: 0 Slot: 0x0 Module ID: 0x1    Subslot: 0x8001 Submodule ID: 0x8001 Index in supported submodules: 2 Dir: 0 In: 0 Out: 0 bytes
Module plug call-back
  Pull old module.    API: 0 Slot: 0x1    Slot was empty.
  Plug module.        API: 0 Slot: 0x1 Module ID: 0x32 Index in supported modules: 3
Submodule plug call-back.
  Pull old submodule. API: 0 Slot: 0x1                   Subslot: 0x1      Subslot was empty.
  Plug submodule.     API: 0 Slot: 0x1 Module ID: 0x32   Subslot: 0x1 Submodule ID: 0x1 Index in supported submodules: 5 Dir: 3 In: 1 Out: 1 bytes
Connect call-back. AREP: 1  Status codes: 0 0 0 0
Callback on event PNET_EVENT_STARTUP
New data status callback. AREP: 1  Status changes: 0x35  Status: 0x35
Parameter write call-back. AREP: 1 API: 0 Slot: 1 Subslot: 1 Index: 123 Sequence: 2 Length: 4
  Bytes: 00 00 00 00 
Parameter write call-back. AREP: 1 API: 0 Slot: 1 Subslot: 1 Index: 124 Sequence: 3 Length: 4
  Bytes: 00 00 00 00 
[ERROR] CMRPC(2520): Unknown packet_type 1
[ERROR] CMRPC(2520): Unknown packet_type 1
Callback on event PNET_EVENT_ABORT. Error class: 253 Error code: 6
Aborting the application
Connect call-back. AREP: 1  Status codes: 0 0 0 0
```
i did not understand why these errors are coming. Some thing did i do wrong. 
For your information, codesys is trial version and "Codesys control for Raspberry Pi SL" not licensed
it runs for 2 hr.
I have attached full logs in the file.

Can you please suggest the reason for errors i saw .

Thanks
Raj

pyhys, 2020-04-09:
Hi!
This seems like a timing issue. Linux (without real-time patches) is not a real-time operating system, and the PLC can be rather sensitive to timing. 

It is possible to adjust the cycle time that the IO-controller (PLC) is using for cyclic data communication with an IO-device. In the left menu of Codesys, double-click the "rt_labs_DEMO_device", and open the "General"tab. The "Communication" section shows the send clock in milliseconds from the GSDML file. By using the "Reduction ratio" you can slow down the communication, by multiplying the cycle time by the factor given in the dropdown.

It is also possible to increase the watchdog time, after which the PLC will set an alarm for missing incoming cyclic data. The watchdog will also shut down the communication, and trigger a subsequent restart of communication.

In case of problems, increase the reduction ratio (and timeout) value a lot, and then gradually reduce it to find the smallest usable value.

Rayapati1203, 2020-04-15:
Hi Jonas Berg,
Thanks for sharing the important tip and sorry for the responding lately.
As per your input, i changed the reduced ratio from 4 to 32  and the corresponding watchdog timeout is changed from 12 to 96. Now the the system looks stable and most of the time it is in Run mode except some glitches i found in the wirshark traffic. Can you have a look at the image and let me know what is the reason it is causing this glitch.
One more thing i observed in IO-device log .. "Error: Out of resources" message . I observed many times this. I am not sure why this is coming.

I have one more question - where can i see the changes in value of 8bit in-8-bit out module connected to slot 1.

[image]

Regards
Raj

pyhys, 2020-04-17:
Hi again!
The parameters for the slot (index 123 and 124) are written once on startup, and will trigger the callback `pnet_write_ind()`. The cyclic data from the PLC are sent every few milliseconds. To read it, use the function `pnet_output_get_data_and_iops()` in your program.

Profinet is a real-time protocol, and the PLC will trigger an alarm if there is too long time between incoming messages. However the topmost alarm (PNIO-AL) seems to be from your IO-device. It is not possible to see the details from your screenshot. The colored lines are the normal start-up sequence messages.

Rayapati1203, 2020-04-17:
Hi 
Thanks for the information . 
I have not captured the full data. But i saw the PNIO-AL packets many times. 
This ticket is not a bug. I am requesting you not to close this tickets for few days.
As mentioned <pnet_output_get_data_and_iops()> , can i use this function to read data and send it to PLC periodically ..can i say every 1 secs?

pyhys, 2020-04-17:
Hi again, 
see https://rt-labs.com/docs/p-net/api_documentation.html#periodic-data on how to use periodic data. The data can be changed every cycle, which typically is a few milliseconds (adjustable from the PLC). 

To read output data from the PLC, use `pnet_output_get_data_and_iops()` To write data to the PLC (= input data in Profinet naming), use  `pnet_input_set_data_and_iops()`.

Rayapati1203, 2020-04-17:
Hi 
Thanks for the information.
I implemented as below to read values from 123 and 124
`pnet_output_get_data_and_iops(net, 
						 APP_API, 
					         slot, 
						 PNET_SUBMOD_CUSTOM_IDENT, 
						 &outputdata_is_updated, 
					         outputdata, 
					         &outputdata_length, 
					         &outputdata_iops);
printf("\n\r slot : [%d] , Subslot : [%d] , Value : [%d] , nof bytes: [%d]\r\n",slot,									    PNET_SUBMOD_CUSTOM_IDENT,outputdata[0],outputdata_length);
`
but the result i got is 
`slot : [1] , Subslot : [1] , Value : [128] , nof bytes: [1]
`
Did i miss any thing here? and Do you have any input to read 123 and 124 values

pyhys, 2020-04-20:
Hi again!
Regarding question1, the parameter value write from the IO-controller will trigger the `pnet_write_ind()` callback. See the sample app.

Question2: You need to write a PLC program to study the incoming data values in the IO-controller. See the page on Codesys in the documentation.

Rayapati1203, 2020-04-20:
Thanks for the information.
Do you have any idea why station status is showing as "DCP no RealStationName" and code is 81813F02. That means the IO device is not detected . I did not see Rasberry pi LLDP packet on Wireshark traffic. The "Codesys Control for Raspberry Pi SL" package i am running in demo mode.  Twodays back i saw in log "it is running demo mode more than 119 mins". Due this the device is not able to talk to IO-Controller? If that is the reason i will take licensed one.

Raj

pyhys, 2020-04-20:
Hi again!
Question1: The parameter values 123 and 124 are set by the IO-controller, and can be read by the IO-controller. When the IO-controller does a read the `pnet_read_ind()` callback will be triggered, and you need to provide the data. When the IO-controller does a write the `pnet_write_ind()` callback will be triggered, and you need to take care about the incoming value. In order for the IO-device to read data from the IO-controller, you need to use the cyclic data. 

The "Codesys Control for Raspberry Pi SL" works the same as the licensed version, but you need to restart it every two hours.

pyhys, 2020-05-05:
Closing due to inactivity. Please reopen if necessary.

closed 2020-05-05 by pyhys