rtlabs-com/p-net #50: Return error value for write request 0xaff1

JanVasina, 2020-04-16:
Hello, I am testing your stack with the Automated RT tester. I have several issues, but for beginning, just some simple:

in the pf_cmrpc.c file, line 1744 shall be

`pf_set_error(p_stat, PNET_ERROR_CODE_WRITE, PNET_ERROR_DECODE_PNIO, PNET_ERROR_CODE_1_CMRPC, PNET_ERROR_CODE_2_CMRPC_AR_UUID_UNKNOWN);`

the error code is PNET_ERROR_CODE_WRITE, not PNET_ERROR_CODE_READ, otherwise the tester complains about invalid data returned. (Testing an invalid write request and the Behavior test 1).

And one typo in the pf_cmdev.c file, line 2766 and 2771, there is two times the condition

`else if (p_exp_sub->data_descriptor[data_ix].length_iops != 1)`

the second one (on line 2771) probably shall be

` else if (p_exp_sub->data_descriptor[data_ix].length_iocs != 1)`

i.e. testing of IOPS and IOCS lengths.

Nevertheless, your stack is an excelent piece of work!

pyhys, 2020-04-16:
Thanks a lot for your detailed report! Any help is appreciated. 
We are in the process of hunting down bugs, and we have been merging quite a number of fixes lately. Any help to locate bugs is very welcome.

I will have a look into it, and will try to provide a fix as soon as possible.

pyhys, 2020-04-17:
Hi again,
I created a pull request to fix the issue. See #53 

Please review, @JanVasina

PR #53 (2020-04-17): Fix error code values sent by p-net
The check of the UUID in incoming write request (in pf_cmrpc.c) is
corrected to Profinet 2.4 section 5.2.7 and 5.2.40.10.2

Check of IOCS field length (in pf_cmdev.c) is corrected to
Profinet 2.4 section 5.2.4.2.5

Closes #50
commit c4b6b328 (2020-04-17): Fix error code values sent by p-net

The check of the UUID in incoming write request (in pf_cmrpc.c) is
corrected to Profinet 2.4 section 5.2.7 and 5.2.40.10.2

Check of IOCS field length (in pf_cmdev.c) is corrected to
Profinet 2.4 section 5.2.4.2.5

Closes #50
closed 2020-04-17 by hefloryd