rtlabs-com/p-net #52: I want to implement a C language service to read data from PLC over Profinet

samminen, 2020-04-16:
I am looking for some help on how to use this library to write a sample C service to connect to PLC (hardware) and read data from Slot-1/Subslot-1/channel.

I want to read this data for every 5 sec from PLC.

Is it possible with this library ?

pyhys, 2020-04-17:
Hi!
You can use this library to implement a Profinet IO-device (a slave) that is talking to a PLC. 
The PLC will control the IO-device, and the data will be exchanged often (on millisecond level).
The IO-device will be written in C, and run on Linux (or rt-kernel). 

Profinet does not use TCP. For the cyclic data it uses plain Ethernet level 2 packets, and UDP is used for the start-up communication. 

To read out data from the PLC itself, i guess it is possible to use other protocols.

Please describe your use case and hardware in some detail.

pyhys, 2020-05-05:
Closing due to inactivity. Please reopen if necessary.

closed 2020-05-05 by pyhys