rtlabs-com/p-net #56: How to run sample app in docker container(SWARM environment)

Rayapati1203, 2020-04-24:
Hi 

I want run the sample app in docker container(SWARM) and configure/connect with PLC. 
When i run the application in Docker i can not use system IP address to configure in PLC because docker container will it own ip address(ex: 10.0.0.1) and this ip address can not be exposed to host machine(as it is protected by docker SWARM).

In general docker services will expose port to host machine for any  communication with host machine.  Since profinet does not have port configured and we should use only ip address to communicate with plc, what are the alternatives to solve this problem?

Thanks for your help aprecited.

Regards
Raj

pyhys, 2020-04-24:
Hi, this seems to be rather Docker-specific, and unfortunately I do not know the answer. 
Maybe you can try to explain the problem in a Docker user forum, and see if you can get some ideas from there?

pyhys, 2020-05-05:
Closing due to inactivity. Please reopen if necessary.

closed 2020-05-05 by pyhys

anildes, 2020-06-09:
One way to do this an adaptor for openvswitch protocol 
Thanks 
Anil Deshpande

pyhys, 2020-06-16:
Thanks Anil for your response!