rtlabs-com/p-net #61: Issue : Station 'rt-labs-dev' : 81813F02-PBNIO: DCP no RealStationName

Rayapati1203, 2020-05-04:
Hi 

I am facing an issue - the device is not discoverable. and i am getting this error.
"Station 'rt-labs-dev' : 81813F02-PBNIO: DCP no RealStationName"

Setup Environment: 
1. Rsapberry pi as PLC with codesys runtime package running.
2. Codesys development
3. p-net stack with sample app is running under VN ubuntu 18.04 LTS

I have captured the issue snap shots for the reference.
1. Codesys service status (on Raspberry pi system)
[image]

2. Raspberrypi Tab : systemInfo capture text:
[Raspberrypi_tab_systemInfo.txt](https://github.com/rtlabs-com/p-net/files/4572515/Raspberrypi_tab_systemInfo.txt)

3. No real Station Capture:
[image]

Earlier it was working and i saw all the network traffic including power-up sequence and cyclic data traffic ..In last week suddenly it started giving this issue. I am not able to understand where was the issue. Do you have any input on this issue?

Regards
Rayapati

Rayapati1203, 2020-05-04:
i found the issue ..My environment is setup all ip addresses for Raspberry pi and virtual box ubuntu linux (device IO) on wifi. The Gateway is different in both raspberry pi and linux environment are different. And Then i changed the ip of raspberry pi and linux to static . Then it started working.
I am closing this thread.

closed 2020-05-04 by Rayapati1203

pyhys, 2020-05-04:
Great, thanks for the update!

ppcolo, 2024-05-04:
> 我发现了问题..我的环境是在 wifi 上为 Raspberry pi 和虚拟盒子 ubuntu linux（设备 IO）设置所有 IP 地址。网关在树莓派和 linux 环境中是不同的。然后我把树莓派和linux的ip改成了静态的。然后它开始工作。我正在关闭这个线程。

我想问下如何设置树莓派的静态ip地址 这个对我来说有点麻烦