rtlabs-com/p-net #69: Include MAC address when creating new UUID

Svenson12, 2020-05-27:
The creation of a new UUID for an outgoing request (like Application Ready) does not create UUIDs that are really unique:
https://github.com/rtlabs-com/p-net/blob/6b83eb308551acd2469c44964de05770c32c18df/src/device/pf_cmrpc.c#L313-L324

As an improvement I would recommend to include at least the MAC address and if possible some timestamp. Additionally the UUID should follow the spec (see https://en.wikipedia.org/wiki/Universally_unique_identifier).

So would recommend something like this:
      p_sess->activity_uuid.data1 = os_get_current_time_us ();
      p_sess->activity_uuid.data2 = net->cmrpc_session_number >> 16;
      p_sess->activity_uuid.data3 = 0x1000 | ((net->cmrpc_session_number >> 8) & 0x00ff);
      p_sess->activity_uuid.data4[0] = 0x80;
      p_sess->activity_uuid.data4[1] = net->cmrpc_session_number & 0xff;
      p_sess->activity_uuid.data4[2] = net->fspm_cfg.eth_addr.addr [0];
      p_sess->activity_uuid.data4[3] = net->fspm_cfg.eth_addr.addr [1];
      p_sess->activity_uuid.data4[4] = net->fspm_cfg.eth_addr.addr [2];
      p_sess->activity_uuid.data4[5] = net->fspm_cfg.eth_addr.addr [3];
      p_sess->activity_uuid.data4[6] = net->fspm_cfg.eth_addr.addr [4];
      p_sess->activity_uuid.data4[7] = net->fspm_cfg.eth_addr.addr [5];
      net->cmrpc_session_number++;

The timestamp does not follow the spec, but the one of my S7 doesn't as well.
With an RTC timestamp you could build an even better UUID, but this could be an issue on some embedded systems.

hefloryd, 2020-05-28:
Regarding RTC we could add a new osal function for getting an RTC timestamp and platforms without RTC could just defer to os_get_current_time_us().

Svenson12, 2020-05-28:
Sounds good...

pyhys, 2020-05-28:
Hi, and thanks for your as always very detailed bug report.

I made a quick implementation in #72 Please review.
It uses the 32 bit timestamp from OSAL. 

@hefloryd If we need a higher resolution RTC timestamp (60 bits needed) how should the OSAL define the type?

hefloryd, 2020-06-02:
Could we not use time_t directly or perhaps abstracting it via the OSAL?

PR #72 (2020-06-04): Make generated UUID actually unique
Closes #69
commit 96d4a7ad (2020-06-04): Make generated UUID actually unique

Drop incoming responses from wrong UUID

Mock UUID generation for repeatable tests

Closes #69
closed 2020-06-04 by hefloryd