rtlabs-com/p-net #73: Not running in CentOS7.7 enviornment

Rayapati1203, 2020-05-28:
Hi 
I am running this app in CentOS7.7 on Virtual box environment. The image i selected as minimal for CentOS.
CentOS:
static address static ip address to 192.168.0.12,255.255.255.0,gateway - 192.168.0.1,inetrface: enp0s3, Bridge adapter
PLC(RASPberry Pi) : static ip address to 192.168.0.10,255.255.255.0,gateway - 192.168.0.1,inetrface: eth0, 
HOST OS (win10) : static ip address to 192.168.0.11,255.255.255.0,gateway - 192.168.0.1

[image]

I am getting the above error.

The same thing i am running successfully in VB Ubuntu 18.04 enviornment. With the same configuration.

When application is running on CentOS and the moment when i start the codesys service on raspberry pi. this error is coming.
Do you have any input on this.

Regards
Raj

pyhys, 2020-05-29:
Hi, you need to adapt the script to fit your Linux distribution. See the help text in https://github.com/rtlabs-com/p-net/blob/master/src/osal/linux/set_network_parameters

pyhys, 2020-06-08:
Closing due to inactivity

closed 2020-06-08 by pyhys