rtlabs-com/p-net #74: Wrong error code for CControl timeout

pyhys, 2020-05-29:
The stack responds with wrong error code if the controller stops communicating. It should be different error codes depending on where in the start-up sequence the error occurs.

Found with Automated RT Tester BEHAVIOR_SCENARIO_5 and BEHAVIOR_SCENARIO_8A

pyhys, 2020-06-08:
For  BEHAVIOR_SCENARIO_5, the PLC fails to respond to CControl. The error value should be:
 p_ar->err_cls = PNET_ERROR_CODE_1_RTA_ERR_CLS_PROTOCOL;
 p_ar->err_code = PNET_ERROR_CODE_2_ABORT_AR_RPC_CONTROL_ERROR;
This is a RPC timeout. Error should be "0xCF, 0x81, 0xFD, 0x18” or “0xCF, 0x81, 0xFD, 0x02”

For  BEHAVIOR_SCENARIO_8, the PLC fails to do more requests. The error value should be:
ErrorCode = RTA-ERR (0xCF), ErrorDecode = PNIO
(0x81), ErrorCode1 = RTA_ERR_CLS_PROTOCOL (253) and ErrorCode2 = AR cmi timeout (06).

atomic-3dpenguin, 2020-06-25:
Confirming that this is fixed via Behavior -  Scenario 1-9 test now passing

commit 83b50258 (2020-07-01): Set proper error code for CControl timeout

Found with Automated RT Tester

Closes #74
closed 2020-07-01 by pyhys