rtlabs-com/p-net #95: Notify user of data value change

pyhys, 2020-05-29:
A request from an end users stand-point is one of the following:

- Have a flag indicating that the data has changed
- Have a callback that triggers an event to inform the application that the data has changed for a given slot/subslot.

Svenson12, 2020-06-12:
I had a simmilar requirement. Here what I added to the stack:
I replaced pf_cpm_put_buf as follows:
`
static void pf_cpm_put_buf(
   pnet_t                  *net,
   pf_iocr_t               *p_iocr,
   os_buf_t                **pp_buf)
{
   pf_cpm_t                *p_cpm = &p_iocr->cpm;
   pf_iodata_object_t      *p_iodata;
   const uint8_t           *p_old_buffer;
   const uint8_t           *p_new_buffer;
   size_t                  ix;
   void                    *p;

   os_mutex_lock(net->cpm_buf_lock);
   p = p_cpm->p_buffer_cpm;
   p_cpm->p_buffer_cpm = *pp_buf;
   *pp_buf = p;

   if ((net->fspm_cfg.change_cb != NULL) &&
       (*pp_buf != NULL))
   {
      /* get pointer to new data buffer */
      p_new_buffer = &((uint8_t*)(p_cpm->p_buffer_cpm->payload))[p_cpm->buffer_pos];

      /* get pointer to old data buffer */
      if ((p_cpm->new_buf == TRUE) || (p_cpm->p_buffer_app == NULL))
      {
         p_old_buffer = &((uint8_t*)((*pp_buf)->payload))[p_cpm->buffer_pos];
      } else
      {
         p_old_buffer = &((uint8_t*)(p_cpm->p_buffer_app->payload))[p_cpm->buffer_pos];
      }

      /* compare content */
      for (ix = 0u; ix < p_iocr->nbr_data_desc; ix++)
      {
         p_iodata = &p_iocr->data_desc[ix];

         if ((p_iodata->in_use == TRUE) &&
             (p_iodata->data_length > 0) &&
             (memcmp (&p_new_buffer[p_iodata->data_offset],
                      &p_old_buffer[p_iodata->data_offset],
                      p_iodata->data_length) != 0))
         {
            /* call callback */
            net->fspm_cfg.change_cb (net, net->fspm_cfg.cb_arg, p_iodata->api_id,
                                     p_iodata->slot_nbr, p_iodata->subslot_nbr);
         }
      }
   }

   p_cpm->new_buf = TRUE;
   os_mutex_unlock(net->cpm_buf_lock);
}
`

In pf_cpm_c_data_ind I call the replaced function now as follows:
`pf_cpm_put_buf(net, p_iocr, &p_buf);`

In pf_cpm_t I redefined p_buffer_app and p_buffer_cpm as os_buf_t* instead void*.

In pnet_cfg_t I added a new call back:
`   pnet_change_ind         change_cb;`
with 
`typedef int (*pnet_change_ind)(
   pnet_t                  *net,
   void                    *arg,
   uint32_t                api,
   uint16_t                slot,
   uint16_t                subslot);
`

In my application I set a flag in the callback function that is handled later on.

Svenson12, 2020-06-12:
An additional cemment: Some callback functions have the parameter api defined as uint16_t, some as uint32_t. I would say, all should be uint32_t.

pyhys, 2020-06-12:
Thanks for your feedback. We will put it in the backlog.

pyhys, 2020-06-12:
Thanks for your very valuable feedback, as always @Svenson12 ! Please review #113