rtlabs-com/p-net #98: UDP sockets are not reliably closed

Svenson12, 2020-06-02:
For the UDP communication with the master sockets will be opened.
When the session is terminated, this socket must be closed.
For this purpose the following 2 lines are implemented in pf_cmrpc_cmdev_state_ind:
https://github.com/rtlabs-com/p-net/blob/cd66dfd317616a2c7cefe65f3e60997bec550e14/src/device/pf_cmrpc.c#L2891-L2892

But sessions are terminated at other points of the code as well and without calling os_udp_close. E.g. in the following lines of pf_cmrpc_dce_packet:
https://github.com/rtlabs-com/p-net/blob/cd66dfd317616a2c7cefe65f3e60997bec550e14/src/device/pf_cmrpc.c#L2751-L2754

In this case the UDP socket will not be closed and all information about this socket will be deleted. As a result the socket will stay opened forever. (And this causes issues on by platform.)

So I recommend to remove the call of os_udp_close from pf_cmrpc_cmdev_state_ind and add it before the following call of memset in pf_session_release:
https://github.com/rtlabs-com/p-net/blob/cd66dfd317616a2c7cefe65f3e60997bec550e14/src/device/pf_cmrpc.c#L351 

The code could be something simple like:
`
         if (p_sess->socket > 0)
         {
            os_udp_close(p_sess->socket);
         }
`

pyhys, 2020-06-02:
Thanks for your report, we will have a look.

pyhys, 2020-06-03:
Your detailed bug reports are very appreciated! 
Please review #100

Svenson12, 2020-06-03:
Looks good. That's exactly what I meant...

PR #100 (2020-06-04): Close UDP socket after use
Closes #98
commit 9b60eaf6 (2020-06-04): Close UDP socket after use

Closes #98
closed 2020-06-04 by hefloryd