rtlabs-com/p-net #109: IN Cent OS The application is not running

Rayapati1203, 2020-06-10:
Hi 
Earlier i  have posted one issue reference -  #73 
Environment :
I am running this app in CentOS7.7 on Virtual box environment. The image i selected as minimal for CentOS.
CentOS:
static address static ip address to 192.168.0.13,255.255.255.0,gateway - 192.168.0.1,inetrface: enp0s3, Bridge adapter
PLC(RASPberry Pi) : static ip address to 192.168.0.10,255.255.255.0,gateway - 192.168.0.1,inetrface: eth0,
HOST OS (win10) : static ip address to 192.168.0.11,255.255.255.0,gateway - 192.168.0.1

IO Modules configuration 
[image]

Now i took the correct script and run the App in centos - i did not get any error wit this script.
snap shot
[image]

Reference Snap shot - codesys IDE with wrieshark image capture
Status of the device: 16C9A049(Host Unreachable_in_device Status)
The device is identified with correct IP and proper station name.  But for sore reason the ARP is not reachable to device(captured the wireshark traffic) . The connection response is not going back to PLC controller from device.
[image]

Do you have any inputs on this scenario, that will be helpful for me. 
I am able to run the application in Ubuntu environment without any issue.

Regards
Rayapati

nakarlsson, 2020-06-21:
Hi @Rayapati1203, can you share your VBox CentOS image?

Rayapati1203, 2020-06-21:
HI 
Can you please provide the Google drive link as you mentioned in the mail.

Regards
Rayapati

nakarlsson, 2020-07-24:
Firewall issue, open ports as described in this link worked. Only open ports to -0xC0FF is not 100% sure, we’ve seen Profinet controllers using port 60000.

https://product-help.schneider-electric.com/Machine%20Expert/V1.1/en/core_ProfinetIO_Configuration_Editor/topics/_pnio_firewall_packagefilter.htm

closed 2020-07-24 by nakarlsson