rtlabs-com/p-net #120: Code Quality Improvements

Svenson12, 2020-06-22:
Here a list of possible code quality improvements:

https://github.com/rtlabs-com/p-net/blob/8d9b8035c5abf96fa1d417abb5d49a6b5d88a10a/src/common/pf_cpm.c#L584
The '&' in front of the last parameter should be removed, simmilar to the call 5 lines above.

https://github.com/rtlabs-com/p-net/blob/8d9b8035c5abf96fa1d417abb5d49a6b5d88a10a/src/common/pf_cpm.c#L741
The index of p_buffer should not be calculated every time. Use 'p_iodata->iops_offset' as index instead.

https://github.com/rtlabs-com/p-net/blob/8d9b8035c5abf96fa1d417abb5d49a6b5d88a10a/src/common/pf_cpm.c#L746
Line should end with ';' instead a comma.

https://github.com/rtlabs-com/p-net/blob/8d9b8035c5abf96fa1d417abb5d49a6b5d88a10a/src/common/pf_dcp.c#L328
Remove '&' in front of 'device_options'.

https://github.com/rtlabs-com/p-net/blob/8d9b8035c5abf96fa1d417abb5d49a6b5d88a10a/src/common/pf_dcp.c#L359-L360
Finish this branch with a 'break' to prevent a fall-through.

https://github.com/rtlabs-com/p-net/blob/8d9b8035c5abf96fa1d417abb5d49a6b5d88a10a/src/common/pf_dcp.c#L412
The '&' in front of the last parameter should be removed.

https://github.com/rtlabs-com/p-net/blob/8d9b8035c5abf96fa1d417abb5d49a6b5d88a10a/src/common/pf_dcp.c#L730
Remove '&' in front of the 2nd parameter.

https://github.com/rtlabs-com/p-net/blob/8d9b8035c5abf96fa1d417abb5d49a6b5d88a10a/src/common/pf_dcp.c#L1300-L1301
Swap '#endif' and 'break;' to prevent unrechable code.

https://github.com/rtlabs-com/p-net/blob/8d9b8035c5abf96fa1d417abb5d49a6b5d88a10a/src/device/pf_block_reader.c#L348
Remove '&' in front of the 4th parameter.

https://github.com/rtlabs-com/p-net/blob/8d9b8035c5abf96fa1d417abb5d49a6b5d88a10a/src/device/pf_block_reader.c#L606
Line should end with just one ';'.

https://github.com/rtlabs-com/p-net/blob/8d9b8035c5abf96fa1d417abb5d49a6b5d88a10a/src/device/pf_block_writer.c#L662
The define 'PNET_SRL_DATA' does not exist. Use 'PNET_OPTION_SRL' instead?

https://github.com/rtlabs-com/p-net/blob/8d9b8035c5abf96fa1d417abb5d49a6b5d88a10a/src/device/pf_block_writer.c#L839
The define 'PNET_SRL_DATA' does not exist. Use 'PNET_OPTION_SRL' instead?
But 'PNET_OPTION_SRL' must be set to 0 or 'srl_data' must be defined.

https://github.com/rtlabs-com/p-net/blob/8d9b8035c5abf96fa1d417abb5d49a6b5d88a10a/src/device/pf_cmdev.c#L710
Line should end with just one ';'.

https://github.com/rtlabs-com/p-net/blob/8d9b8035c5abf96fa1d417abb5d49a6b5d88a10a/src/device/pf_cmina.c#L522
Invalid type of last parameter. Should be 'PNET_EVENT_ABORT' instead.

https://github.com/rtlabs-com/p-net/blob/8d9b8035c5abf96fa1d417abb5d49a6b5d88a10a/src/device/pf_cmina.c#L544
Invalid type of last parameter. Should be 'PNET_EVENT_ABORT' instead.

https://github.com/rtlabs-com/p-net/blob/8d9b8035c5abf96fa1d417abb5d49a6b5d88a10a/src/device/pf_cmina.c#L641
Remove '&' in front of '...device_vendor'.

https://github.com/rtlabs-com/p-net/blob/8d9b8035c5abf96fa1d417abb5d49a6b5d88a10a/src/device/pf_cmsu.c#L111-L112
Remove one space in front of 'if' for clear indention.

pyhys, 2020-06-23:
Thanks for your detailed feedback!

Please review #122

Svenson12, 2020-06-23:
Looks good for me...

PR #122 (2020-06-23): Code quality improvement
Closes #120
commit 655bbb37 (2020-06-23): Code quality improvement

Closes #120
closed 2020-06-23 by hefloryd