rtlabs-com/p-net #147: Segmentation fault in pf_alarm.c

JanVasina, 2020-07-22:
File **pf_alarm.c**:

The function `pf_alarm_apms_a_data_req()` has as its fourth parameter `pf_alarm_data_t  *p_alarm_data`, which is then used to store alarm data by `pf_put_alarm_block()` function. This parameter is never checked against NULL.
But the `pf_alarm_apms_a_data_req()` is called with its **p_alarm_data** parameter set to NULL in the `pf_alarm_alpmr_alarm_ack()` function. 
This subsequently leads to segmentation fault (i.e. dereferencing NULL pointer in the `pf_put_alarm_block()` function).

Somewhere a check against NULL is necessary, either in the `pf_alarm_apms_a_data_req()` or maybe in the low-level function `pf_put_alarm_block()`.

pyhys, 2020-08-06:
Hi, thanks for your report.
Yes you are right.

I think I need to change the public API for `pnet_alarm_ind()` and `pnet_alarm_send_ack()` to have the needed info to populate `p_alarm_data`.

These functions are used when the IO-Controller (PLC) sends an alarm. 
Do you have any suggestions on how to test this in a PLC? The PLC will for example send an alarm when it detects timeout, but is there some other way to trigger some alarm? Programmatically? Any suggestions on how to do this on Siemens or Codesys would be great.

JanVasina, 2020-08-07:
I am not a big expert in Profinet, neither in TIA Portal. All my efforts is just to adapt the p-net stack source code for our device to pass all the tests necessary for Profinet IO device certification (now with success :-)).

The Automated RT Tester itself sends alarms in the "Alarm" test. Maybe you can test the alarms this way?

pyhys, 2020-08-12:
Hi,
please review #153 @JanVasina !

PR #153 (2020-09-17): Additional data for public alarm ack API
Improved transport ACK (TACK) handling.
Corrected writing of outgoing messages.
Closes #147
commit 84ef315e (2020-09-17): Improved transport ACK (TACK) handling.
Corrected writing of outgoing messages.

Closes #147
closed 2020-09-17 by hefloryd