rtlabs-com/p-net #198: arduino support ?

yanzixiang, 2020-10-19:
is there any plan to support arduino ?

pyhys, 2020-10-20:
No, unfortunately not. The memory requirements are probably too high.

yanzixiang, 2020-10-20:
If you have the time,
you can have a look at this chip,
https://www.espressif.com/en/products/socs/esp32/resources
and the arduino support at here
https://github.com/espressif/arduino-esp32

betodr4, 2021-01-13:
Hi @yanzixiang .
I am trying to use this p-net library on esp32 board.
Do you have some examples for esp32 board??

pyhys, 2021-04-29:
Closing due to inactivity.
Please reopen if necessary.

closed 2021-04-29 by pyhys

yanzixiang, 2021-07-12:
@pyhys
https://item.taobao.com/item.htm?spm=a230r.1.14.6.2097329e3WTurb&id=625821996883&ns=1&abbucket=14#detail
please have a look at this.