rtlabs-com/p-net #288: Find the right function.

spedro, 2020-12-30:
Dear all, I am completely new to the profinet world, so please forgive me (and correct) if I write inaccuracies or errors.

I have to retrieve data from an application, run on a laser scanner, which "speaks in profinet". 

The application is already in use and therefore reachable via IP through a layer 2 switch, all the network configuration has already been done and working. 

So, I would like to wrote a small piece of code that, executed on a Linux system, will be able to make the proper profinet request to the application and obtain data back.
 
At each request, the application should return me 3 values (say x, y, z). 

Is there, in the p-net library, a function that helps me to get this information ?

In addition to the ip address to query, what other parameters should I know to make such a request ?

Many thanks for the help.

pyhys, 2021-01-04:
Hi, and thanks for your interest in p-net!
In Profinet you have IO-devices and IO-controllers. An IO-controller is also known as a PLC (programmable logic controller). The p-net library is for implementing an IO-device. See more details on https://rt-labs.com/docs/p-net/profinet_basics.html

My guess is that the laser scanner acts as an IO-device, so you need to implement a IO-controller. Unfortunately the p-net library can not act as an IO-controller.

Please verify that the laser scanner acts as an IO-device.

pyhys, 2021-02-19:
Closed due to inactivity. Please reopen if necessary.

closed 2021-02-19 by pyhys