rtlabs-com/p-net #315: Correct shutdown sequence

giuseppe-avon, 2021-01-26:
I am using the library for my slave device. I am experiencing troubles when I run two consecutive (non-parallel) runs: while the first goes without problems, the second one never receives the State Indication change and causes a "00006001 RPC_Request_Blocked" on the Codesys Control for Linux instance. I am issuing the pnet_ar_abort command to then wait for the Abort State Indication, after that I just stop calling the pnet_handle_periodic(). What I don't really understand is if this is the right way to handle the slave disconnection, leaving things clean for the next instance to be executed. Thanks.

giuseppe-avon, 2021-02-08:
Is there something which may be dual to the pnet_init, in order to close the AF_PACKET socket left open?

giuseppe-avon, 2021-02-09:
FWIW apparently solved by just adding
`setsockopt(id, SOL_SOCKET, SO_REUSEADDR, &one, sizeof(one));`

just before the
`ret = bind (id, (struct sockaddr *)&local, sizeof (local));`

in function
`int pnal_udp_open (pnal_ipaddr_t addr, pnal_ipport_t port)`

Seems like stray sockets remain when closing happens by means of interruption. However, this pinpoints to a misbehaving close/cleanup stage.

pyhys, 2021-02-09:
Thanks for your update!
We need look into this.

Best regards
Jonas

PR #374 (2021-03-17): Set SO_REUSEADDR for UDP sockets to enable fast restarts
Closes #315
commit 0fe63030 (2021-03-17): Set SO_REUSEADDR for UDP sockets to enable fast restarts

Closes #315
closed 2021-03-17 by hefloryd