rtlabs-com/p-net #323: Compilation issues

giuseppe-avon, 2021-02-03:
Hi
Did a pull today from the repository and this is what I obtain when issuing the cmake command

CMake Error at CMakeLists.txt:181 (add_executable):
  Target "pf_test" links to target "Threads::Threads" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?

CMake Error at CMakeLists.txt:178 (add_executable):
  Target "pn_dev" links to target "Threads::Threads" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?

CMake Error at CMakeLists.txt:177 (add_library):
  Target "profinet" links to target "Threads::Threads" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?

Dependecies are met and also tried using "internal" and "external" OSAL, as suggested on the documentation.
Thanks in advance for your advices

giuseppe-avon, 2021-02-04:
Just adding to the Linux.cmake, at appropriate position

find_package(Threads REQUIRED)

target_link_libraries(profinet
  PUBLIC
  Threads::Threads
  rt
  $<$<BOOL:${PNET_OPTION_SNMP}>:NetSNMP::NetSNMPAgent>
  $<$<BOOL:${PNET_OPTION_SNMP}>:NetSNMP::NetSNMP>
  INTERFACE
  $<$<CONFIG:Coverage>:--coverage>
  )

seems solving it

hefloryd, 2021-02-04:
Hi Guiseppe,

Could you let us know which version of osal you are building against? There are some updates to osal that use Threads::Threads but the version used in p-net (44e5a39, see AddOsal.cmake) does not use them yet.

Also please post complete build logs (including command line and output of cmake configuration stage).

FWIW, automated builds work fine, see for instance https://travis-ci.com/github/rtlabs-com/p-net/builds/216044462.

giuseppe-avon, 2021-02-04:
Vanilla Minimal CentOS7, + Development Tools
gcc version 4.8.5 20150623 (Red Hat 4.8.5-44) (GCC)
cmake3 version 3.17.5

```
[user@localhost p-net]$ git branch -vv
* master 96c6bdf [origin/master] Git should not do line end normalization on bash scripts

```

```
[user@localhost profinet]$ cmake3 -B build -S p-net -DBUILD_SHARED_LIBS=ON
-- The C compiler identification is GNU 4.8.5
-- The CXX compiler identification is GNU 4.8.5
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Current build type is: RelWithDebInfo
-- Current install path is: /home/user/profinet/build/install
-- Building for Linux
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY - Success
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY - Success
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
-- Found PythonInterp: /usr/bin/python (found version "2.7.5")
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.5") found components: doxygen missing components: dot
-- Configuring done
CMake Error at CMakeLists.txt:181 (add_executable):
  Target "pf_test" links to target "Threads::Threads" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?

CMake Error at CMakeLists.txt:181 (add_executable):
  Target "pf_test" links to target "Threads::Threads" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?

CMake Error at CMakeLists.txt:178 (add_executable):
  Target "pn_dev" links to target "Threads::Threads" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?

CMake Error at CMakeLists.txt:177 (add_library):
  Target "profinet" links to target "Threads::Threads" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?

CMake Error at CMakeLists.txt:177 (add_library):
  Target "profinet" links to target "Threads::Threads" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
```

hefloryd, 2021-02-04:
I am not able to reproduce this in a clean Centos7 container:

```
[root@centos7 ~]# cd p-net/
[root@centos7 p-net]# uname -a
Linux centos7 5.4.0-65-generic #73-Ubuntu SMP Mon Jan 18 17:25:17 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
[root@centos7 p-net]# cmake --version
cmake version 3.17.5

CMake suite maintained and supported by Kitware (kitware.com/cmake).
[root@centos7 p-net]# git describe --tags --always --dirty 
b276cf7
[root@centos7 p-net]# cmake -B build -S . -DBUILD_SHARED_LIBS=ON
-- The C compiler identification is GNU 4.8.5
-- The CXX compiler identification is GNU 4.8.5
-- Check for working C compiler: /bin/cc
-- Check for working C compiler: /bin/cc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /bin/c++
-- Check for working CXX compiler: /bin/c++ - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Current build type is: RelWithDebInfo
-- Building for Linux
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- Current build type is: RelWithDebInfo
-- Current install path is: /root/p-net/build/install
-- Building for Linux
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY - Success
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY - Success
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
-- Found PythonInterp: /bin/python (found version "2.7.5") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE  
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- Configuring done
-- Generating done
-- Build files have been written to: /root/p-net/build

```

Perhaps you could zip up your build folder and post it here? The cmake logs should have some clues.

giuseppe-avon, 2021-02-08:
Seen inside AddOsal and got the right branch for it, as you were pointing out. The thing is that the auto-pulling from the Osal doesn't happen on my side so I gave up, manually pulled Osal at your branch and followed the tutorial as in "external Osal". However thanks for pointing me in the right direction.

closed 2021-02-09 by hefloryd

jmachowinski, 2022-04-27:
Hi,
this bug is still present.
To reproduce it, you need to have a version of osal compiled and installed.
Configure p-net to the same install destination and you will hit it.

The cause of this error is, that  the CMakeLists.txt in osal generates an
osalConfig.cmake that contains
` INTERFACE_LINK_LIBRARIES "Threads::Threads;rt;\$<\$<CONFIG:Coverage>:--coverage>"`
As the osalConfig does not include a findPackage(Threads), this generates the error.

We got a fix for this, should we open a merge request ?

hefloryd, 2022-04-27:
Yes, there is a problem consuming an installed osal as the generated config is missing the required find_package, as you say. Your fix looks good, see my comments there.