rtlabs-com/p-net #329: Files inclusion error while compilation

Ap040, 2021-02-05:
I just cloned the repository and try to compile it on Ubuntu 20.04. I encountered the problem in CMake to include "AddOsal", "AddGoogleTest" and "add_gtest". Can someone help so, I can build it successfully.

> CMake Error at CMakeLists.txt:21 (include):
> include could not find load file:
> 
> AddOsal
> -- Current build type is: RelWithDebInfo
> -- Current install path is: /mnt/e/Local/Downloads/PN_Device_Linux/p-net/build/install
> -- Building for Linux
> CMake Error at CMakeLists.txt:229 (include):
> include could not find load file:
> 
> AddGoogleTest
> CMake Error at CMakeLists.txt:230 (add_gtest):
> Unknown CMake command "add_gtest".
> 
> -- Configuring incomplete, errors occurred!

hefloryd, 2021-02-08:
Did you clone with submodules? If not, try running:

`git submodule update --init --recursive`

Also see https://rt-labs.com/docs/p-net/tutorial.html#id1.

Ap040, 2021-02-09:
Thank You so much @hefloryd hefloryd ...!!!
I clone it as per your comment and it is working. Also, it is compiled successfully and generated the **_pn_dev_** file.
Will try the remaining steps to run the application and connect with the PN-Master.

closed 2021-02-09 by hefloryd