rtlabs-com/p-net #392: Codesys doesn't recognize my raspberry Pi

Gaetanos-99, 2021-03-30:
Hi hi,

i'm currently setting an IO-Device (Raspberry Pi) and a IO-Controller (Windows Laptop). I connected the my Raspi through a USB Serial Cable to my Laptop. I did everything in the tutorial (GSDML file uploaded in codeSys, uart on my Raspi enabled, etc.). But I have the 2 following issues:
* First issue: I can find my Rapi in codesys:
[image]

But when I choose the network interface, there is the ethernet interface eth0, which has everything setted as 0:
[image]

* As soon as I call the command "sudo ifconfig eth0 192.168.0.50 netmask 255.255.255.0 up" in order to set my ethernet interface or as soon as I run the command "./pn_dev -v", my Raspi doesn't appear in codesys (Device -> Search Network) anymore.

Can someone please help me? I am quite new in programming Profinet. Thank you in advance

pyhys, 2021-04-01:
Hi!
You need two Raspberry Pi:s. One running as an IO-device and one running as an IO-controller. Your Windows laptop is used to make changes to the IO-controller settings, but it does not act as an IO-controlller.

Best regards
Jonas

Gaetanos-99, 2021-04-02:
Thank you very much for your reply! Is there a way to use my laptop as IO-Controller so that the stack would still function?

Best regards
Gaetan

pyhys, 2021-04-07:
Hi again!
The only IO-controller variants that I know of are a physical PLC (like Siemens) or running the Codesys software on a Raspberry Pi, unfortunately.

Best regards
Jonas

Gaetanos-99, 2021-04-07:
Hi,
thank you very much for your comment! I'll now close it.

Ap040, 2021-04-07:
Hello,
I want to know the same thing.
If we test the sample application, can we do it with two PCs?. Like one PC has run the CODESYS PN-master and the other has run our sample application. Both PCs should have connected with an Ethernet cable.

Can this scenario works?.

pyhys, 2021-04-07:
Best @Ap040!
Not to my knowledge, unfortunately not.

Best regards
Jonas

Gaetanos-99, 2021-04-28:
Hi again!
Now I am using two raspberry Pis but I still have the same issue. On my Raspberry Pi as an IO Controller the eth0 is still 0. Can you please help me?

Best regards
Gaetan

Gaetanos-99, 2021-04-28:
https://github.com/rtlabs-com/p-net/issues/416

closed 2021-04-28 by Gaetanos-99