rtlabs-com/p-net #396: Endpont mapper (EPM) request does not release session

pyhys, 2021-04-07:
Multiple EPM requests will use up all available sessions.

pmgreen346, 2021-04-07:
Steps to reproduce with TIA Portal:
- Open TIA Portal
- On the Portal View, go to Online & Diagnostics -> Accessible devices
- In the Accessible devices dialog, click the Start Search button
   - TIA will send an Ident Request followed by an EPM Lookup request

[image]

- Wait for the search to complete and click Start Search again
- Do this for PF_MAX_SESSION + 1 times and the CMRPC sessions will be exhausted.

The resulting log and traffic capture is attached.  When the failure occurs you see TIA sending Ping requests due to lack of response to the EPM request in the traffic, and "out of session resources" logs in the log.

More details:
- Each EPM Lookup request sent by TIA uses a different Activity UUID, resulting in a new session
- It appears that the only way that sessions are released in the stack is by a Release or Abort event, neither of which are ever sent in this scenario.

[Repeated_EPM_Requests.pcap.zip](https://github.com/rtlabs-com/p-net/files/6271212/Repeated_EPM_Requests.pcap.zip)
[Repeated_EPM_Requests_log.txt](https://github.com/rtlabs-com/p-net/files/6271213/Repeated_EPM_Requests_log.txt)

pyhys, 2021-04-07:
Great!
I will have a look.

PR #400 (2021-04-13): Close EPM session after each request
TIA portal uses EPM requests for identifying devices on the network.

Previously the sessions were not closed, resulting in "Out of session resources"
when a number of requests have been done.

Profinet uses EPM in a limited way. If we for some reason we need more advanced
EPM handling in the future, implement a timeout for each EPM session.

Closes #396
commit 4d066ccc (2021-04-13): Close EPM session after each request

TIA portal uses EPM requests for identifying devices on the network.

Previously the sessions were not closed, resulting in "Out of session resources"
when a number of requests have been done.

Profinet uses EPM in a limited way. If we for some reason we need more advanced
EPM handling in the future, implement a timeout for each EPM session.

Closes #396
closed 2021-04-13 by hefloryd