rtlabs-com/p-net #399: cross compiler

wildwindyevan, 2021-04-12:
I will run that code on imx 6ul platform .It is pure linux 。
When I tried to cmake ,  I was told that need OSAL.
So give me some  suggestion for  cross compiler "p-net" on arm linux

pyhys, 2021-04-12:
Hi!
Please see https://rt-labs.com/docs/p-net/tutorial.html#advanced-users-osal on how to download OSAL.

Best regards
Jonas

wildwindyevan, 2021-04-12:
Thanks for your reply .Another  mistake happened :
================
_CMake Error at CMakeLists.txt:49 (include):
  include could not find load file:

    GetGitRevision_
================

pyhys, 2021-04-12:
Hi!
Are you using the latest version?
Have you done:
`git submodule update --init --recursive`

Best regards
Jonas

pyhys, 2021-04-21:
Closing due to inactivity. Please reopen if necessary.

closed 2021-04-21 by pyhys