rtlabs-com/p-net #416: Network interface et0 doesn't receive any packets

Gaetanos-99, 2021-04-27:
Hi hi,
first of all, thank you for providing such a wonderful stack! It is extremely useful.
I followed everything on this tutorial. But I bump on the following problem: The network interface eth0 of my raspberry pi (as an IO Controller) isn't configured in the Codesys Software. Everything is 0 when I choose eth0 as network configuration.
[image]

I already included in "/etc/dhcpcd.conf" the lines: "interface eth0" and "static ip_address=192.168.0.100/24". Can someone please help me? I'm really stuck with this...

pyhys, 2021-04-29:
Hi!
You can run the command `ifconfig` on your Raspberry Pi to show the settings of the network interfaces. Please provide the output.

If you have problems getting Codesys up an running, there is lots of info at https://forge.codesys.com/forge/talk/

Best regards
Jonas

Gaetanos-99, 2021-05-04:
Hi Jonas,
thank you for your reply. Here is the output:
[image]

Here is my configuration in /etc/dhcpcd.conf:
[image]

Best regards
Gaetan

Gaetanos-99, 2021-05-26:
I had to connect my Raspi to the other Raspi with an Ethernet cable and it worked. I also had to define eth0 ip adresses of both Raspis in another range. Now it's functionning.

Best regards
Gaetan

closed 2021-05-26 by Gaetanos-99