rtlabs-com/p-net #424: LED usage in sampleapp

illishar, 2021-05-07:
Your sampleapp has 2 LEDs. The signal LED (fair enough) and a APP_DATA_LED_ID.

The APP_DATA_LED_ID is only connected to false. (Through the app_set_outputs_default_value.)

What was your intent, with this?

pyhys, 2021-05-11:
The data LED is set via
```
static void app_handle_data_led_state (bool led_state, int verbosity)
{
   static bool previous_led_state = false;

   if (led_state != previous_led_state)
   {
      app_set_led (APP_DATA_LED_ID, led_state, verbosity);
   }
   previous_led_state = led_state;
}
```
That function is called from `app_set_outputs_default_value()` but also from `app_handle_cyclic_data()`.

Best regards
Jonas

illishar, 2021-05-12:
Ah, it's a debug feature, it seems. Thank you.

closed 2021-05-12 by illishar