rtlabs-com/p-net #430: Assign PLC variables to Raspberry Pi pins?

Gaetanos-99, 2021-05-26:
Hi,

I was trying to understand this part in the documentation (Section: Using Codesys soft PLC):
"On the “DIO_8xLogicLevel” node in the left-side menu, right-click and select “Edit IO mapping”. Open the “Input 8 bits” row by clicking the small + sign. Double-click the icon on the row that you would like the edit. Map “Input Bit 7” (found via Application/PLC_PRG) to “in_pin_button_LED”, and “Output Bit 7” to “out_pin_LED”.". 

My LED that is turning on, when pressing Button1 (GPIO27), is on the Pin 11 (GPIO17). 
My question is: how is "Input Bit 7" somehow related to GPIO27? Why not Input Bit 6 or 5? I have the same question also for Output Bit 7. Thank you in advance for your answers.

Best regards
Gaetan

pyhys, 2021-05-27:
Hi!
We have arbitrarily decided that our sample app should receive one byte (8 bits) of (output-)data from the PLC, and that the most significant bit should control a LED on the IO-device.
There are two LEDs in the sample app, and this LED is the "data LED" or "LED1". See https://rt-labs.com/docs/p-net/tutorial.html#sample-app-description

In` app_data_set_output_data()` in `sample_app/app_data.c` we extract that bit from the outputdata:
```
led_state = (outputdata[0] & 0x80) > 0;
```
So we now know whether the LED should be on or off.  

For Linux, that information is sent to `app_set_led()` in `src/ports/linux/sampleapp_main.c` That function calls a script with the arguments which LED (1 or 2) and whether it should be on or off.
The script itself can be found in `src/ports/linux/set_profinet_leds.raspberrypi`
It contains:
```
if  [ ${LED_NUMBER} == "1" ]; then
   GPIO=17
elif [ ${LED_NUMBER} == "2" ]; then
   GPIO=23
```
Modify that script if you would like to change GPIO pins.

Similar mechanism applies for reading the buttons.

Best regards
Jonas

pyhys, 2021-06-21:
Closing due to inactivity. Please reopen if necessary.

closed 2021-06-21 by pyhys

Gaetanos-99, 2021-07-23:
Thanks a lot for your answer! Where can I have find the files/methods reponsible for reading the buttons? 

Best regards
Gaetan