rtlabs-com/p-net #440: Bug in pf_cmdev_iocr_setup_iocs function

JanVasina, 2021-09-17:
Hello, I have found a bug in the function pf_cmdev_iocr_setup_iocs() in the pf_cmdev.c source file. There is a call of pf_cmdev_get_exp_sub() function in the loop, but the second parameter is wrong -> it should be **api_id**, not _api_ix_.

So instead of (starting at line 2129 of pf_cmdev.c)

```
         api_id = p_iocr_param->apis[api_ix].api;
         slot_nbr = p_iocr_param->apis[api_ix].iocs[ix].slot_number;
         subslot_nbr = p_iocr_param->apis[api_ix].iocs[ix].subslot_number;
         if (
            pf_cmdev_get_exp_sub (
               p_ar,
               api_ix, // <- wrong parameter here
               slot_nbr,
               subslot_nbr,
               &p_exp_sub) != 0)
         {

```
there should be written

```
         api_id = p_iocr_param->apis[api_ix].api;
         slot_nbr = p_iocr_param->apis[api_ix].iocs[ix].slot_number;
         subslot_nbr = p_iocr_param->apis[api_ix].iocs[ix].subslot_number;
         if (
            pf_cmdev_get_exp_sub (
               p_ar,
               api_id,   // <- here is the change!!
               slot_nbr,
               subslot_nbr,
               &p_exp_sub) != 0)
         {

```

May be it would address (or solve?) also the #issue #438 .

_Just a small note about the coding style:_
The AUTOSAR (or MISRA) set of coding rules does **NOT** recommend to use names of variables (or functions, etc.) which differs only in **one** letter. I am sure that the above "bug" is a typo only, but leads to non-function code where there is more than one API. (Note that with the API=0 and only one API, the api_ix is equal to api_id (=0) and the code works as expected.)

If the names of variables inside the functions would differ more, the probability of such a typo would be much lower. :-)

pyhys, 2021-09-20:
Great,
thanks for the report!

JanVasina, 2021-09-29:
Just a note: the same wrong **api_ix** variable is assigned in the function pf_cmdev_iocr_setup_iocs() on line 2205. Instead of

```
iodata_cnt++;

p_iodata->api_id = api_ix; // <-- this is wrong
p_iodata->slot_nbr = slot_nbr;
p_iodata->subslot_nbr = subslot_nbr;
```

there should be

```
iodata_cnt++;

p_iodata->api_id = api_id; // <-- here is the change
p_iodata->slot_nbr = slot_nbr;
p_iodata->subslot_nbr = subslot_nbr;

```

PR #449 (2021-12-06): Use api_id instead of api_ix
Bug reported by Jan Vasina

Closes #440
commit f09193e1 (2021-12-06): Use api_id instead of api_ix

Bug reported by Jan Vasina

Closes #440
closed 2021-12-06 by hefloryd