rtlabs-com/p-net #442: Unable to build after latest commit

bft-automation-team, 2021-09-21:
Dear all,
After pulling the latest commit i'm not able to build the sample app anymore.
This is the error log:

```
Scanning dependencies of target gtest
[  0%] Building CXX object _deps/googletest-build/googletest/CMakeFiles/gtest.dir/src/gtest-all.o
[  1%] Linking CXX static library ../../../lib/libgtest.a
[  1%] Built target gtest
Scanning dependencies of target osal
[  2%] Building C object _deps/osal-build/CMakeFiles/osal.dir/src/linux/osal.o
[  3%] Building C object _deps/osal-build/CMakeFiles/osal.dir/src/linux/osal_log.o
[  4%] Linking C static library libosal.a
[  4%] Built target osal
Scanning dependencies of target profinet
[  5%] Building C object CMakeFiles/profinet.dir/src/ports/linux/pnal.o
[  6%] Building C object CMakeFiles/profinet.dir/src/ports/linux/pnal_eth.o
[  7%] Building C object CMakeFiles/profinet.dir/src/ports/linux/pnal_udp.o
[  8%] Building C object CMakeFiles/profinet.dir/src/ports/linux/pnal_filetools.o
[  9%] Building C object CMakeFiles/profinet.dir/src/device/pnet_api.o
[  9%] Building C object CMakeFiles/profinet.dir/src/device/pf_block_reader.o
[ 10%] Building C object CMakeFiles/profinet.dir/src/device/pf_block_writer.o
[ 11%] Building C object CMakeFiles/profinet.dir/src/device/pf_fspm.o
[ 12%] Building C object CMakeFiles/profinet.dir/src/device/pf_diag.o
[ 13%] Building C object CMakeFiles/profinet.dir/src/device/pf_cmdev.o
[ 14%] Building C object CMakeFiles/profinet.dir/src/device/pf_cmdmc.o
[ 15%] Building C object CMakeFiles/profinet.dir/src/device/pf_cmina.o
[ 16%] Building C object CMakeFiles/profinet.dir/src/device/pf_cmio.o
[ 17%] Building C object CMakeFiles/profinet.dir/src/device/pf_cmpbe.o
[ 18%] Building C object CMakeFiles/profinet.dir/src/device/pf_cmrdr.o
[ 19%] Building C object CMakeFiles/profinet.dir/src/device/pf_cmrpc.o
[ 20%] Building C object CMakeFiles/profinet.dir/src/device/pf_cmrpc_epm.o
[ 21%] Building C object CMakeFiles/profinet.dir/src/device/pf_cmrpc_helpers.o
[ 22%] Building C object CMakeFiles/profinet.dir/src/device/pf_cmrs.o
[ 23%] Building C object CMakeFiles/profinet.dir/src/device/pf_cmsm.o
[ 23%] Building C object CMakeFiles/profinet.dir/src/device/pf_cmsu.o
[ 24%] Building C object CMakeFiles/profinet.dir/src/device/pf_cmwrr.o
[ 25%] Building C object CMakeFiles/profinet.dir/src/device/pf_pdport.o
[ 26%] Building C object CMakeFiles/profinet.dir/src/device/pf_port.o
[ 27%] Building C object CMakeFiles/profinet.dir/src/common/pf_alarm.o
[ 28%] Building C object CMakeFiles/profinet.dir/src/common/pf_bg_worker.o
[ 29%] Building C object CMakeFiles/profinet.dir/src/common/pf_cpm.o
[ 30%] Building C object CMakeFiles/profinet.dir/src/common/pf_dcp.o
[ 31%] Building C object CMakeFiles/profinet.dir/src/common/pf_ppm.o
[ 32%] Building C object CMakeFiles/profinet.dir/src/common/pf_ptcp.o
[ 33%] Building C object CMakeFiles/profinet.dir/src/common/pf_scheduler.o
[ 34%] Building C object CMakeFiles/profinet.dir/src/common/pf_eth.o
[ 35%] Building C object CMakeFiles/profinet.dir/src/common/pf_file.o
[ 36%] Building C object CMakeFiles/profinet.dir/src/common/pf_lldp.o
[ 37%] Building C object CMakeFiles/profinet.dir/src/common/pf_udp.o
[ 38%] Linking C static library libprofinet.a
[ 38%] Built target profinet
Scanning dependencies of target pf_test
[ 39%] Building CXX object CMakeFiles/pf_test.dir/test/test_alarm.o
[ 40%] Building CXX object CMakeFiles/pf_test.dir/test/test_block_reader.o
[ 41%] Building CXX object CMakeFiles/pf_test.dir/test/test_cmdev.o
[ 42%] Building CXX object CMakeFiles/pf_test.dir/test/test_cmdmc.o
[ 43%] Building CXX object CMakeFiles/pf_test.dir/test/test_cmina.o
[ 44%] Building CXX object CMakeFiles/pf_test.dir/test/test_cmio.o
[ 45%] Building CXX object CMakeFiles/pf_test.dir/test/test_cmpbe.o
[ 46%] Building CXX object CMakeFiles/pf_test.dir/test/test_cmrdr.o
[ 46%] Building CXX object CMakeFiles/pf_test.dir/test/test_cmrpc.o
[ 47%] Building CXX object CMakeFiles/pf_test.dir/test/test_cmrs.o
[ 48%] Building CXX object CMakeFiles/pf_test.dir/test/test_cmsm.o
[ 50%] Building CXX object CMakeFiles/pf_test.dir/test/test_cmsu.o
[ 51%] Building CXX object CMakeFiles/pf_test.dir/test/test_cmwdr.o
[ 52%] Building CXX object CMakeFiles/pf_test.dir/test/test_cpm.o
[ 53%] Building CXX object CMakeFiles/pf_test.dir/test/test_dcp.o
[ 54%] Building CXX object CMakeFiles/pf_test.dir/test/test_diag.o
[ 55%] Building CXX object CMakeFiles/pf_test.dir/test/test_eth.o
[ 56%] Building CXX object CMakeFiles/pf_test.dir/test/test_file.o
[ 57%] Building CXX object CMakeFiles/pf_test.dir/test/test_fspm.o
[ 58%] Building CXX object CMakeFiles/pf_test.dir/test/test_lldp.o
[ 59%] Building CXX object CMakeFiles/pf_test.dir/test/test_pnetapi.o
[ 60%] Building CXX object CMakeFiles/pf_test.dir/test/test_port.o
[ 61%] Building CXX object CMakeFiles/pf_test.dir/test/test_ppm.o
[ 61%] Building CXX object CMakeFiles/pf_test.dir/test/test_ptcp.o
[ 62%] Building CXX object CMakeFiles/pf_test.dir/test/test_scheduler.o
[ 63%] Building CXX object CMakeFiles/pf_test.dir/test/utils_for_testing.o
[ 64%] Building CXX object CMakeFiles/pf_test.dir/test/mocks.o
[ 65%] Building CXX object CMakeFiles/pf_test.dir/test/pf_test.o
[ 66%] Building C object CMakeFiles/pf_test.dir/src/device/pf_block_reader.o
[ 67%] Building C object CMakeFiles/pf_test.dir/src/device/pf_block_writer.o
[ 68%] Building C object CMakeFiles/pf_test.dir/src/device/pf_fspm.o
[ 69%] Building C object CMakeFiles/pf_test.dir/src/device/pf_diag.o
[ 70%] Building C object CMakeFiles/pf_test.dir/src/device/pf_cmdev.o
[ 71%] Building C object CMakeFiles/pf_test.dir/src/device/pf_cmdmc.o
[ 72%] Building C object CMakeFiles/pf_test.dir/src/device/pf_cmina.o
[ 73%] Building C object CMakeFiles/pf_test.dir/src/device/pf_cmio.o
[ 74%] Building C object CMakeFiles/pf_test.dir/src/device/pf_cmpbe.o
[ 75%] Building C object CMakeFiles/pf_test.dir/src/device/pf_cmrdr.o
[ 75%] Building C object CMakeFiles/pf_test.dir/src/device/pf_cmrpc.o
[ 76%] Building C object CMakeFiles/pf_test.dir/src/device/pf_cmrpc_helpers.o
[ 77%] Building C object CMakeFiles/pf_test.dir/src/device/pf_cmrs.o
[ 78%] Building C object CMakeFiles/pf_test.dir/src/device/pf_cmsm.o
[ 79%] Building C object CMakeFiles/pf_test.dir/src/device/pf_cmsu.o
[ 80%] Building C object CMakeFiles/pf_test.dir/src/device/pf_cmwrr.o
[ 81%] Building C object CMakeFiles/pf_test.dir/src/device/pf_pdport.o
[ 82%] Building C object CMakeFiles/pf_test.dir/src/device/pnet_api.o
[ 83%] Building C object CMakeFiles/pf_test.dir/src/common/pf_alarm.o
[ 84%] Building C object CMakeFiles/pf_test.dir/src/common/pf_cpm.o
[ 85%] Building C object CMakeFiles/pf_test.dir/src/common/pf_dcp.o
[ 86%] Building C object CMakeFiles/pf_test.dir/src/common/pf_eth.o
[ 87%] Building C object CMakeFiles/pf_test.dir/src/common/pf_lldp.o
[ 88%] Building C object CMakeFiles/pf_test.dir/src/common/pf_file.o
[ 89%] Building C object CMakeFiles/pf_test.dir/src/common/pf_ppm.o
[ 90%] Building C object CMakeFiles/pf_test.dir/src/common/pf_ptcp.o
[ 90%] Building C object CMakeFiles/pf_test.dir/src/common/pf_scheduler.o
[ 91%] Building C object CMakeFiles/pf_test.dir/src/common/pf_udp.o
[ 92%] Linking CXX executable pf_test
[ 92%] Built target pf_test
Scanning dependencies of target pn_dev
[ 93%] Building C object CMakeFiles/pn_dev.dir/sample_app/sampleapp_common.o
In file included from /home/plexus/profinet/p-net/sample_app/sampleapp_common.c:18:
/home/plexus/profinet/p-net/sample_app/app_gsdml.h:248:26: error: ‘app_gsdml_parameters’ defined but not used [-Werror=unused-variable]
  248 | static app_gsdml_param_t app_gsdml_parameters[] = {
      |                          ^~~~~~~~~~~~~~~~~~~~
/home/plexus/profinet/p-net/sample_app/app_gsdml.h:229:38: error: ‘app_gsdml_submodules’ defined but not used [-Werror=unused-variable]
  229 | static const app_gsdml_submodule_t * app_gsdml_submodules[] = {
      |                                      ^~~~~~~~~~~~~~~~~~~~
/home/plexus/profinet/p-net/sample_app/app_gsdml.h:225:35: error: ‘app_gsdml_modules’ defined but not used [-Werror=unused-variable]
  225 | static const app_gsdml_module_t * app_gsdml_modules[] =
      |                                   ^~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[2]: *** [CMakeFiles/pn_dev.dir/build.make:63: CMakeFiles/pn_dev.dir/sample_app/sampleapp_common.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:286: CMakeFiles/pn_dev.dir/all] Error 2
make: *** [Makefile:163: all] Error 2
```

pyhys, 2021-09-22:
Thanks, we will have a look.

hefloryd, 2021-09-22:
Are you sure you are running the latest commit? The error you are seeing should have been fixed by fdf14f36.

bft-automation-team, 2021-09-22:
Hi @hefloryd, I followed the tutorial 1 week ago with flawlessly, then again yesterday on another machine and the build failed.
If I switch to the source code I downloaded 1 week ago, everything is OK.
The OS is Ubuntu 20.04 for both.

hefloryd, 2021-09-22:
Please verify that you have commit fdf14f3 in your working copy. You'll notice that in that commit, app_gsdml_parameters was moved from app_gsdml.h to app_gsdml.c, whereas the error message you posted above indicates that app_gsdml_parameters is still being defined in app_gsdml.h.

FWIW, all PR:s are automatically built before merging. You can see the output of the automated build [here](https://github.com/rtlabs-com/p-net/runs/3632913192), it successfully built the latest commit from the unmodified sources.

bft-automation-team, 2021-09-22:
Dear @hefloryd  I don't have the commit, even if I cloned the repo yesterday.

bft-automation-team, 2021-09-22:
I can see that commit now in the output

```
20b0dcf (HEAD -> master, origin/master, origin/HEAD) Update documentation, mostly about the MRP protocol
fdf14f3 Fix sample program warnings
8ca756e Enable warnings for sample program
0f3fcd4 Improve documentation week 22
d755841 Fix assignment of led_state in sample application
71c198a Set log level info for rt-kernel sample app
b7a90e0 Add background worker
ad5147c Close session after incoming RPC CANCEL, FAULT and REJECT messages.
d0ccfe7 Fix USE_SCHED_FIFO option
f603f45 Bugfix RPC max response length
a42cd18 Add rpc fragmentation log message
07f3a5f Refactor sample application
c7344eb Add log message if cycle time is violated
7eca054 Improve documentation week17
d0ac782 Show details for ports and SNMP
4b61d51 Activate both alarm output queues
86d3595 Use unbuffered file reading on Linux
db3c1ee Simplify usage of the scheduler
738565c Keep SNMP values also in memory to avoid file reading
cfdaeb9 Move relevant LLDP types to pf_lldp.h
41de8b2 Improve documentation week 16
365a31a Log file access times
```

bft-automation-team, 2021-09-22:
sorry, i had that commit in my working dir

bft-automation-team, 2021-09-22:
as
[image]
d

hefloryd, 2021-09-23:
I'm not sure from your response if this builds for you now. If not, please post the output of:

```
git describe --always --dirty
cmake -B build.new
cmake --build build.new 
```
Also note that the commit you have highlighted in your previous screenshot would produce the error you are seeing. The subsequent commit fixes those errors.

bft-automation-team, 2021-09-23:
Sorry for wasting your time.
Everything works, thank you for your support and your work

closed 2021-09-23 by bft-automation-team