rtlabs-com/p-net #453: Assigning module from PLC

Und5hadow, 2021-12-15:
Hi,

I want to communicate the sample app with a PLC.
The app is at the moment running on a ubuntu 18.04. It is recognized by the controller, but when I want to add a module on the sample app through the controller nothing happens. 
So I decided to force the plug of a module on startup of the application. But when I read the sample app configuration from the controller, I can't see the added module. Same thing happen when I "push" the added module on the sample app from the controller.

I checked on Wireshark and I found that the device is not responding to the connect request of the controller.
[image]

pyhys, 2022-02-25:
Have you got the sample app running, when following the instructions in detail?
If not, what exact error messages are you seeing?

Best regards
Jonas

pyhys, 2022-03-28:
Closing due to inactivity. 

Please reopen if not solved.

closed 2022-03-28 by pyhys