rtlabs-com/p-net #478: GSD FILE

Tahe97, 2022-03-22:
Good afternoon , 
i tried to use the gsd file in this project to configure communication between RPI 4 and s7 1500 .after configuration s7 doesn't reconized RPI as a device . 
any advices ?

pyhys, 2022-03-23:
You can use Proneta to scan the network for Profinet devices. Verify that your (RPI-based) IO-device show up there. 
I would recommend that as the first step. 
See https://rt-labs.com/docs/p-net/network_topology_detection.html#siemens-proneta-profinet-network-analysis-tool

Tahe97, 2022-03-23:
I tried to install Protena on RPI but i think that i have a problem in the connection WIFI so i tried to install Protena on windows but also he doesn't reconized it it shows only plc .
for the hardware : i use the two ports ethernet of the plc one for the pc and one for the raspberry 
sorry for dump question but i m new in automatisme and i m gratefull for you help

pyhys, 2022-03-24:
Try to follow the tutorial as close as possible, and run with DEBUG log level. Paste the printout from the sample app here, and I will try to help you.

Tahe97, 2022-03-25:
good morning 
i tried to folow the tuto but i have such commands don work this is what i have .
for the lldp it is working good 
`lldpcli show neighbors 
-------------------------------------------------------------------------------
LLDP neighbors:
-------------------------------------------------------------------------------
Interface:    eth0, via: LLDP, RID: 9, Time: 0 day, 00:00:07
  Chassis:     
    ChassisID:    local S7-1500                   6ES7 513-1AL02-0AB0  S C-K5PE09262018     1 V  2  5  0
    SysDescr:     Siemens, SIMATIC S7, CPU 1513-1 PN, 6ES7 513-1AL02-0AB0, HW: Version 1, FW: Version V2.5.0, S C-K5PE09262018
    MgmtIP:       192.168.0.1
    Capability:   Bridge, on
    Capability:   Station, on
  Port:        
    PortID:       local port-002.plcxb2d1ad
    PortDescr:    Siemens, SIMATIC S7, Ethernet Port, X1 P2R
    TTL:          20
  Unknown TLVs:
    TLV:          OUI: 00,0E,CF, SubType: 1, Len: 20 00,00,01,39,00,00,00,00,00,00,00,29,00,00,00,00,00,00,00,00
    TLV:          OUI: 00,0E,CF, SubType: 2, Len: 4 00,00,00,00
    TLV:          OUI: 00,0E,CF, SubType: 5, Len: 6 AC,64,17,13,FB,59
-------------------------------------------------------------------------------
Interface:    wlan0, via: LLDP, RID: 8, Time: 0 day, 00:27:18
  Chassis:     
    ChassisID:    local jackie
    SysName:      JACKIE
    SysDescr:     Micro-Star International Co., Ltd. GF63 Thin 9RCX,REV:1.0,9S716R312678ZJ8000115 + HMI
    MgmtIP:       192.168.137.1
    Capability:   Station, on
  Port:        
    PortID:       local port-001
    TTL:          20
  Unknown TLVs:
    TLV:          OUI: 00,0E,CF, SubType: 2, Len: 4 00,00,00,00
    TLV:          OUI: 00,0E,CF, SubType: 5, Len: 6 0E,DD,24,A7,89,65
-------------------------------------------------------------------------------`
but for the snmp i have a problem with the version v2c
`snmpwalk -v2c -c public 192.168.0.1
Timeout: No Response from 192.168.0.1`
but it'is working with v1
`snmpwalk -v1 -c public 192.168.0.1
iso.3.6.1.2.1.1.1.0 = STRING: "Siemens, SIMATIC S7, CPU 1513-1 PN, 6ES7 513-1AL02-0AB0, HW: Version 1, FW: Version V2.5.0, S C-K5PE09262018"
iso.3.6.1.2.1.1.2.0 = OID: ccitt.0
iso.3.6.1.2.1.1.3.0 = Timeticks: (606910) 1:41:09.10
iso.3.6.1.2.1.1.4.0 = ""
iso.3.6.1.2.1.1.5.0 = ""
iso.3.6.1.2.1.1.6.0 = ""
iso.3.6.1.2.1.1.7.0 = INTEGER: 78
iso.3.6.1.2.1.1.8.0 = Timeticks: (0) 0:00:00.00
iso.3.6.1.2.1.1.9.1.2.1 = OID: iso.0.8802.1.1.2
iso.3.6.1.2.1.1.9.1.2.2 = OID: iso.0.62439.1
iso.3.6.1.2.1.1.9.1.2.3 = OID: iso.3.6.1.2.1
iso.3.6.1.2.1.1.9.1.2.4 = OID: iso.3.6.1.2.1.4
iso.3.6.1.2.1.1.9.1.2.5 = OID: iso.3.6.1.2.1.5
iso.3.6.1.2.1.1.9.1.2.6 = OID: iso.3.6.1.2.1.6
iso.3.6.1.2.1.1.9.1.2.7 = OID: iso.3.6.1.2.1.7
iso.3.6.1.2.1.1.9.1.2.8 = OID: iso.3.6.1.2.1.11
iso.3.6.1.2.1.1.9.1.2.9 = OID: iso.3.6.1.2.1.31
iso.3.6.1.2.1.1.9.1.3.1 = STRING: "iso.0.iso8802.ieee802dot1.ieee802dot1mibs.lldp"
iso.3.6.1.2.1.1.9.1.3.2 = STRING: "iso.0.iec62439.mrp"
iso.3.6.1.2.1.1.9.1.3.3 = STRING: "iso.org.dod.internet.mgmt.mib_2"
iso.3.6.1.2.1.1.9.1.3.4 = STRING: "iso.org.dod.internet.mgmt.mib_2.ipMIB"
iso.3.6.1.2.1.1.9.1.3.5 = STRING: "iso.org.dod.internet.mgmt.mib_2.icmpMIB"
iso.3.6.1.2.1.1.9.1.3.6 = STRING: "iso.org.dod.internet.mgmt.mib_2.tcpMIB"
iso.3.6.1.2.1.1.9.1.3.7 = STRING: "iso.org.dod.internet.mgmt.mib_2.udpMIB"
iso.3.6.1.2.1.1.9.1.3.8 = STRING: "iso.org.dod.internet.mgmt.mib_2.snmp"
iso.3.6.1.2.1.1.9.1.3.9 = STRING: "iso.org.dod.internet.mgmt.mib_2.ifMIB"
iso.3.6.1.2.1.1.9.1.4.1 = Timeticks: (0) 0:00:00.00
iso.3.6.1.2.1.1.9.1.4.2 = Timeticks: (0) 0:00:00.00
[... 170 lines cut ...]
iso.3.6.1.2.1.11.13.0 = Counter32: 199
iso.3.6.1.2.1.11.14.0 = Counter32: 0
iso.3.6.1.2.1.11.15.0 = Counter32: 0
iso.3.6.1.2.1.11.16.0 = Counter32: 202
iso.3.6.1.2.1.11.17.0 = Counter32: 0
iso.3.6.1.2.1.11.18.0 = Counter32: 0
iso.3.6.1.2.1.11.19.0 = Counter32: 0
iso.3.6.1.2.1.11.20.0 = Counter32: 0
iso.3.6.1.2.1.11.21.0 = Counter32: 0
iso.3.6.1.2.1.11.22.0 = Counter32: 0
iso.3.6.1.2.1.11.24.0 = Counter32: 0
iso.3.6.1.2.1.11.25.0 = Counter32: 0
iso.3.6.1.2.1.11.26.0 = Counter32: 0
iso.3.6.1.2.1.11.27.0 = Counter32: 0
iso.3.6.1.2.1.11.28.0 = Counter32: 0
iso.3.6.1.2.1.11.29.0 = Counter32: 0
iso.3.6.1.2.1.11.30.0 = INTEGER: 2
iso.3.6.1.2.1.11.31.0 = Counter32: 0
iso.3.6.1.2.1.11.32.0 = Counter32: 0
End of MIB
`
also i cant do any snmpset 
`snmpset -v1 -c private 192.168.0.1 1.3.6.1.2.1.1.6.0 "My new location"
1.3.6.1.2.1.1.6.0: Bad object type: M
`

Tahe97, 2022-03-25:
Also to be informed this what i have in the plc it doesn't reconized the raspberry 
[image]
[image]

pyhys, 2022-03-25:
Could you please provide a printout from the sample app?

Tahe97, 2022-03-29:
honestly i didn't understand you verry well and i really nead your healp because i have made many researshs i didn't find any thing helpfull . this is my address mail  you can send me tahe.taamallah97@gmail.com i will expalin to you what i want to do and what i have done . thank you

Tahe97, 2022-03-30:
it worked very well. i didn't understand you from the first about the sample application . but it's ok thank uou sir

closed 2022-03-30 by Tahe97