rtlabs-com/p-net #483: stm32

Tahe97, 2022-04-04:
good morning Sir ,
trying to understand this exemple i find some folders about stm32 so i'm asking if we use cmake with the stm32 IDE do you think that will work with stm32 board ?

pyhys, 2022-04-05:
Hi!
Please see the newly updated documentation on https://rt-labs.com/docs/p-net/getting_started_freertos.html

Tahe97, 2022-04-05:
ahh okaay Sir i will try it and give you the result

Tahe97, 2022-04-05:
i tried to do this tuto but i think that i have a problem in this two commands : 
`DCMAKE_TOOLCHAIN_FILE=cmake/tools/toolchain/stm32cube.cmake \`
`G "Unix Makefiles"`
I tried to do every command independly 
i m confused about CUBE_DIR =" /path/to/cube " i put the path of the placement where i clonned the STM32Cube firmware

pyhys, 2022-04-06:
What kind of error message do you get?

Tahe97, 2022-04-06:
it s to long but this what i get in the first command 
`-DCMAKE_TOOLCHAIN_FILE=cmake/tools/toolchain/stm32cube.cmake
`
the result -->
pi@raspberrypi:~/stm32/p-net $ CPU=cortex-m7fd BOARD=STM32F767ZI-NUCLEO CUBE_DIR=/home/pi/stm32/STM32CubeF7 cmake \
>  -DCMAKE_TOOLCHAIN_FILE=cmake/tools/toolchain/stm32cube.cmake
CMake Warning:
  No source or binary directory provided.  Both will be assumed to be the
  same as the current working directory, but note that this warning will
  become a fatal error in future CMake releases.

-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- The ASM compiler identification is unknown
-- Didn't find assembler
CMake Error at cmake/tools/Platform/STM32Cube.cmake:18 (enable_language):
  No CMAKE_ASM_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "ASM" or the CMake cache entry CMAKE_ASM_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/CMakeSystemSpecificInformation.cmake:26 (include)
  CMakeLists.txt:19 (project)

-- Warning: Did not find file Compiler/-ASM
CMake Error at CMakeLists.txt:19 (project):
  The CMAKE_C_COMPILER:

    arm-none-eabi-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:19 (project):
  The CMAKE_CXX_COMPILER:

    arm-none-eabi-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Configuring incomplete, errors occurred!
See also "/home/pi/stm32/p-net/CMakeFiles/CMakeOutput.log".
See also "/home/pi/stm32/p-net/CMakeFiles/CMakeError.log".

Tahe97, 2022-04-06:
the same msg for the second command also 
`CPU=cortex-m7fd BOARD=STM32F767ZI-NUCLEO CUBE_DIR=/home/pi/stm32/STM32CubeF7 cmake \
> -G "Unix Makefiles"
`

Tahe97, 2022-04-06:
the same msg for the second command also
`CPU=cortex-m7fd BOARD=STM32F767ZI-NUCLEO CUBE_DIR=/home/pi/stm32/STM32CubeF7 cmake `

`-G Unix Makefiles`

pyhys, 2022-04-06:
Note that it is just two commands. The first is:
```
CPU=cortex-m7fd BOARD=STM32F769I-DISCO CUBE_DIR=/path/to/cube cmake -B build.cube -DCMAKE_TOOLCHAIN_FILE=cmake/tools/toolchain/stm32cube.cmake -G "Unix Makefiles"
```
and the seconds is 
```
cmake --build build.cube`
```
Note that cmake is called in both lines. In the first are three environmental variables set before executing cmake.

Tahe97, 2022-04-06:
Yes that s exactly what i have done . for the first command 

> CPU=cortex-m7fd BOARD=STM32F769I-DISCO CUBE_DIR=/path/to/cube cmake -B build.cube -DCMAKE_TOOLCHAIN_FILE=cmake/tools/toolchain/stm32cube.cmake -G "Unix Makefiles"
for the first environmental variable it s ok but for the second and the third i have the msg that i have sent .  that s exactly my problem 

for the value of the `CUBE_DIR` i put this directory ``

Tahe97, 2022-04-06:
I have solved this problem it was in the `CUBE_DIR` path but unfortunately after building i didn't find `build.cube/pn_dev.bin` only an executable file named pn_dev (type : XMLspy)

pyhys, 2022-04-08:
Which operating system do you run the commands on?
Do you get any error messages?
Have you tried loading the file to your board?

Tahe97, 2022-04-08:
* i'm working on raspberry pi 4 .
* i have the error that i have printed in this msg 

> _https://github.com/rtlabs-com/p-net/issues/483#issuecomment-1090109741_

* yes i tried to change the name of pn_dev (type : xmlspy) on pn_dev.bin and i upload it but i have no changes unfortunately .

pyhys, 2022-04-08:
It seems that you are missing the ARM cross compiler. Install it with:

`sudo apt install gcc-arm-none-eabi`

I will improve the documentation in future releases.

Tahe97, 2022-04-08:
I tried to install virtual machine and working on it is that ok ?

Tahe97, 2022-04-27:
good morning pyhys ,
i'm still working on the same project, i tried to apply the existing code on STM32'IDE . i made some changes and the code compiled without errors but i have encountered a problem in  sample_app file mainly in the `app_loop_forever module, there weren't any verified coditions. i would be grateful if you could help

SolarPeng2022, 2022-05-01:
An Exsample Build In Keil or IAR for STM32 is the best

andre-lost-a-pig, 2023-02-09:
I build p-net in Keil5. Feel free to get it.
[p-net-stm32](https://github.com/andre-lost-a-pig/p-net-stm32)

pyhys, 2024-04-23:
Closing due to inactivity. Please reopen if necessary.

closed 2024-04-23 by pyhys