rtlabs-com/p-net #493: [ERROR] CMRPC(4775): Unknown packet_type 7

hangeV, 2022-05-01:
我们打算用本协议栈做复杂设备，现在测试发现一个问题：一旦添加超过40个带Record的module就无法组态成功。协议栈提示该错误"[ERROR] CMRPC(4775): Unknown packet_type 7"。你们又进行过这种测试吗

pyhys, 2022-05-02:
Please provide a Wireshark file, and I will have a look at the problem.

hangeV, 2022-05-02:
> 请提供一个 Wireshark 文件，我会看看问题所在。
[image]

我们用STM32裸机开发。UDP协议是自己写的，没有做IP层分包合并处理，UDP最大能收发1472个字节(1514-14-20-8)，我不知道会不会和这里有关系。1200PLC和设备通过一根网线直连，没有经过路由器，按说也不会发生IP分包的情况。

hangeV, 2022-05-02:
[image]
[image]
我们用STM32裸机开发。UDP是我们自己写的，没有IP层拆包合并处理，可以收发(1514-14-20-8)个字节，我不知道会不会和这里有关系。1200PLC和我们开发板通过网线直连，没有经过路由器，没有跨网关，应该不会发生IP分包的情况。报错的是这个函数。

hangeV, 2022-05-02:
> 请提供一个 Wireshark 文件，我会看看问题所在。

插入超过40个带Record的module会无法组态失败，会"APP_EVENT_ABORT"。而不带Record的则超过40个也不会有问题。
在sampleap_common.c的app_loop_forever()函数

hangeV, 2022-05-05:
> 

现在大概找到了问题的所在，但是我不会解决。
[image]
我发现当"PF_MAX_UDP_PAYLOAD_SIZE"变小，则能组态成功的数量越少。比如，这个宏定义为1440时我可以组态40个带Record的module，当改为1472时可以组态41个，当改为800时只能组态不到20个。我打印了PF_RPC_LOG日志，里面有三部分，第一个部分是组态成功时的日志，后两个部分是两个一直组态失败的日志。
moduls的Record长度是3(除了固定在第一个槽位的Module的Record长度是8)。
问题可能是出在"pf_cmrpc_dce_packet"函数，可能是"case: PF_RPC_PT_FRAG_ACK :"
**************************************************************************************************************************************
//SUCCESS    //SUCCESS    //SUCCESS    //SUCCESS    //SUCCESS    //SUCCESS    //SUCCESS    //SUCCESS    //SUCCESS    //SUCCESS    //SUCCESS    //SUCCESS    //SUCCESS    //SUCCESS    //SUCCESS    //SUCCESS    //SUCCESS    //SUCCESS    
[13:59:25.996]收←◆r: 1464
      3282 [INFO ] CMRPC(4922): Received 1464 bytes UDP payload from remote 192.168.0.1:49153, on socket 1 for incoming DCE RPC requests.
      3284 [DEBUG] CMRPC(4147): Incoming RPC frame. Seq: 0, Serial: 0, Frag num: 0, Frag Len: 1384 Fragment flag: "Set" Last Fragment flag: "Not set", No fack: "Not set"
      3286 [DEBUG] CMRPC(527): Allocated session 0
      3286 [INFO ] CMRPC(4789): Send fragment ACK
      3287 [INFO ] CMRPC(914): Sending 80 bytes on socket 1 to 192.168.0.1:49153 Payload:"fragment ack" Session from me:0 Session index:0
s: 80
r: 964
      3290 [INFO ] CMRPC(4922): Received 964 bytes UDP payload from remote 192.168.0.1:49153, on socket 1 for incoming DCE RPC requests.
      3292 [DEBUG] CMRPC(4147): Incoming RPC frame. Seq: 0, Serial: 1, Frag num: 1, Frag Len: 884 Fragment flag: "Set" Last Fragment flag: "Set", No fack: "Set"
      3293 [INFO ] CMRPC(4350): Incoming DCE RPC request on UDP.
      3294 [INFO ] CMRPC(3487): Incoming CONNECT request via DCE RPC on UDP
      3295 [DEBUG] CMRPC(696): Allocate AR 0 (AREP 1)
      3296 [DEBUG] CMRPC(1202): AR type: "Controller AR" Device access: 0 Supervisor takeover: 0 Requested start up mode: "Advanced" Initiator station name: "plcxb1d0ed" UDP port: 0x8892 Session: 255 Timeout: 200 x 100 ms
      3298 [DEBUG] CMRPC(1233): Requested send cycle time: 32 (in 1/32 of millisec) AREP 1 CREP 0  INPUT Reduction ratio: 2 Watchdog factor: 255 Data hold factor: 255 FrameID: 0x8000
      3300 [DEBUG] CMRPC(1233): Requested send cycle time: 32 (in 1/32 of millisec) AREP 1 CREP 1 OUTPUT Reduction ratio: 2 Watchdog factor: 255 Data hold factor: 255 FrameID: 0x8001
      3302 [DEBUG] CMRPC(1399): Requested max alarm data length: 256 bytes
      3303 [DEBUG] CMRPC(1576): Connect request AR param: 1 CR: 2 Input: 1 Output: 1 Alarm: 1
      3307 [DEBUG] CMRPC(4997): Received event PNET_EVENT_STARTUP from CMDEV for AREP 1.
      3308 [DEBUG] CMRPC(1947): Create CONNECT response: ret = 0   error: 0x00 0x00 0x00 0x00
      3309 [DEBUG] CMRPC(1790): Connect response args_length = 90
      3310 [DEBUG] CMRPC(1974): Created connect response: ret 0
      3311 [DEBUG] CMRPC(4498): Send RPC response. Total response length 190, sending 190 bytes. Start of RPC payload: 80
      3312 [INFO ] CMRPC(914): Sending 190 bytes on socket 1 to 192.168.0.1:49153 Payload:"response" Session from me:0 Session index:0
s: 190
r: 1464
      3323 [INFO ] CMRPC(4922): Received 1464 bytes UDP payload from remote 192.168.0.1:49153, on socket 1 for incoming DCE RPC requests.
      3324 [DEBUG] CMRPC(4147): Incoming RPC frame. Seq: 1, Serial: 0, Frag num: 0, Frag Len: 1384 Fragment flag: "Set" Last Fragment flag: "Not set", No fack: "Not set"
      3326 [INFO ] CMRPC(4789): Send fragment ACK
      3327 [INFO ] CMRPC(914): Sending 80 bytes on socket 1 to 192.168.0.1:49153 Payload:"fragment ack" Session from me:0 Session index:0
s: 80
r: 1464
      3332 [INFO ] CMRPC(4922): Received 1464 bytes UDP payload from remote 192.168.0.1:49153, on socket 1 for incoming DCE RPC requests.
      3333 [DEBUG] CMRPC(4147): Incoming RPC frame. Seq: 1, Serial: 1, Frag num: 1, Frag Len: 1384 Fragment flag: "Set" Last Fragment flag: "Not set", No fack: "Not set"
      3335 [INFO ] CMRPC(4789): Send fragment ACK
      3336 [INFO ] CMRPC(914): Sending 80 bytes on socket 1 to 192.168.0.1:49153 Payload:"fragment ack" Session from me:0 Session index:0
s: 80
r: 195
      3339 [INFO ] CMRPC(4922): Received 195 bytes UDP payload from remote 192.168.0.1:49153, on socket 1 for incoming DCE RPC requests.
      3340 [DEBUG] CMRPC(4147): Incoming RPC frame. Seq: 1, Serial: 2, Frag num: 2, Frag Len: 115 Fragment flag: "Set" Last Fragment flag: "Set", No fack: "Set"
      3342 [INFO ] CMRPC(4350): Incoming DCE RPC request on UDP.
      3343 [INFO ] CMRPC(3528): Incoming WRITE request via DCE RPC on UDP
      3345 [DEBUG] CMRPC(4498): Send RPC response. Total response length 2788, sending 1440 bytes. Start of RPC payload: 80
      3347 [INFO ] CMRPC(914): Sending 1440 bytes on socket 1 to 192.168.0.1:49153 Payload:"response fragment" Session from me:0 Session index:0
s: 1440
r: 100
      3350 [INFO ] CMRPC(4922): Received 100 bytes UDP payload from remote 192.168.0.1:49153, on socket 1 for incoming DCE RPC requests.
      3351 [DEBUG] CMRPC(4147): Incoming RPC frame. Seq: 1, Serial: 3, Frag num: 0, Frag Len: 20 Fragment flag: "Not set" Last Fragment flag: "Not set", No fack: "Not set"
      3353 [DEBUG] CMRPC(4538): Received fragment ACK.
      3354 [DEBUG] CMRPC(4550): More fragments to send. Total RPC payload 2788, already sent 1440, max fragment UDP payload 1440 (incl RPC header). args_length 2688
      3356 [DEBUG] CMRPC(4595): Sending last fragment.
[... cut ...]
      4451 [DEBUG] CMRPC(4642): Send RPC fragment. Total response length 3556, already sent 2880, sending 676 bytes RPC payload. RPC payload start pos 80, Fragment size: 756 bytes
      4453 [INFO ] CMRPC(914): Sending 756 bytes on socket 1 to 192.168.0.1:49153 Payload:"response fragment" Session from me:0 Session index:1
s: 756
      4457 [DEBUG] CMRPC(4997): Received event PNET_EVENT_ABORT from CMDEV for AREP 1.
      4457 [DEBUG] CMRPC(565): Released session 1
      4458 [DEBUG] CMRPC(718): Free AR 0 (AREP 1)

hangeV, 2022-05-11:
第三包丢了开头80个字节

hangeV, 2022-05-12:
[wireshark.zip](https://github.com/rtlabs-com/p-net/files/8674299/wireshark.zip)

pyhys, 2023-06-01:
Closing due to inactivity

closed 2023-06-01 by pyhys