rtlabs-com/p-net #502: 1500PLC超过19个带Recod的slot就无法组态成功，1200PLC超过40个带Record的slot就无法组态成功。

hangeV, 2022-06-02:
1500PLC超过19个带Recod的slot就无法组态成功，1200PLC超过40个带Record的slot就无法组态成功。
wireshark抓取1500PLC分别插入10、20、40个带Record的slot。
我认为问题在pf_cmrpc.c文件的pf_cmrpc_dce_packet()函数。
因为之前用1200PLC超过40个带Record的slot无法组态成功，我通过如下修改解决了。
[image]

https://github.com/rtlabs-com/p-net/issues/493。
但现在用1500PLC测试又出现了超过19个带Record的slot也会无法组态成功，无论我有没有按图片上修改pf_cmrpc_dce_packet()函数都是同样的结果。
当然我图片上所示的修改肯定不解决"19"的问题，以为"20"不足以分三包，而"41"会分三包。

hangeV, 2022-06-03:
通过wireshark抓包发现：1500PLC"write request"是“little endian”，两包(当超过18个带Record的slot会分包)都是"little endian"。而p-net"write response"时第一包是"big endian"，而第二包是"little endian"。所以组态失败。

pyhys, 2023-06-01:
Please provide an english translation.

closed 2024-02-21 by pyhys