rtlabs-com/p-net #513: support MIPS architecture （openwrt）

SupermanOfHeiLinPu, 2022-07-27:
the os is openwrt

rich1111, 2022-09-05:
Hi, i've build for MT7620A MIPS architecture by using OpenWRT's toolchain (all building via Jetbrain CLion tool). It is sccessfully built and running, and can discovery/configuration via Profinet Test tool. But while testing with Codesys PLC system, the Profitnet IO reporting errors. The Profinet controller reporting:

2022-09-05T08:06:38.484Z: Cmp=Profinet Controller, Class=1, Error=33554432, Info=0, pszInfo= Station 'rt-labs-dev': Connecting...
2022-09-05T08:06:38.493Z: Cmp=Profinet Controller, Class=1, Error=33554432, Info=0, pszInfo= Station 'rt-labs-dev': Received Connect Confirmation
2022-09-05T08:06:38.528Z: Cmp=Profinet Controller, Class=4, Error=50331648, Info=0, pszInfo=**** Station 'rt-labs-dev': CF81FD0B - Aborted: AR alarm.ind(err)

And by Wireshark package with "pn_rt" filter, it has following PROFINET IO Alarm:

Frame 3941: 56 bytes on wire (448 bits), 56 bytes captured (448 bits) on interface \Device\NPF_{D787D171-8ED3-4E24-9CD6-786E2213AE34}, id 0
Ethernet II, Src: 1VergeIn_29:c2:00 (54:36:9b:29:c2:00), Dst: AsixElec_f5:4e:f9 (00:0e:c6:f5:4e:f9)
PROFINET acyclic Real-Time, Alarm Low, ID:0xfe01, Len:  40
    FrameID: 0xfe01 (Real-Time: Acyclic PN-IO Alarm low priority)
PROFINET IO Alarm
    AlarmDstEndpoint: 0x0001
    AlarmSrcEndpoint: 0x0001
    PDUType, Type: ERR-RTA-PDU, Version: 1
    AddFlags, Window Size: 1, Tack: 0
    SendSeqNum: 0xffff
    AckSeqNum: 0xfffe
    VarPartLen: 0x0004
    Status: Error: "RTA error", "PNIO", "RTA_ERR_CLS_PROTOCOL", "PDEV, no port offers required speed/duplexity (RTA_ERROR_ABORT)"

Dose it means that the MT7620A is not capable for Profinet IO RT ?

SupermanOfHeiLinPu, 2022-09-13:
Maybe the Watchdog time is too small. Or the network in system not right. (Sorry, I'm a green hand. Maybe not right.)

SupermanOfHeiLinPu, 2022-09-13:
But in my branch, I can use it with codesys. Before I run this program, I will run the cmd "ethmode l". I use MT7628N.

branch:feature/wrt
https://github.com/SupermanOfHeiLinPu/p-net.git

rich1111, 2022-09-13:
Hi Superman,
Thank you for your response.
I've test your feature/wrt branch, it work very well.
But how to test your device with your  GSD file? I use rt-labs's PLC_PRG sample.
I would like to implement the IO device just like rt-labs's DIO sample, so I need to modify rt-labs's GSD file?

rich1111, 2022-09-20:
Hi Superman,

Can you let rt-labs's DIO sample work on MT76xx ?

SupermanOfHeiLinPu, 2022-09-20:
In this bran I use another GSD file .It also in the branch.
https://github.com/SupermanOfHeiLinPu/p-net/blob/feature/wrt/samples/pn_dev/GSDML-V2.4-Helium-Flash-P-Net-IO-Station-20220324.xml

SupermanOfHeiLinPu, 2022-09-20:
yes, the sample can work on MT7628N.

rich1111, 2022-09-20:
Hi Supperman,
Thank you for your response. I've try your GSD, but it can not add DIO/DI/DO devices.

SupermanOfHeiLinPu, 2022-10-11:
Hi,
You can use rt-labs's GSD file without modify. But I modify the sample source code in my branch.

rich1111, 2022-11-23:
https://github.com/rich1111/p-net