rtlabs-com/p-net #529: Wrong Outputdata Length

ibndias, 2022-12-22:
Hi, I am new to Profinet. And tried following the demo's instructions [here](https://rt-labs.com/docs/p-net/tutorial.html).
I have set up both raspberry pi as a controller (`raspiprofinetA/192.168.68`) and one as IO (`raspiprofinetB/192/168.0.69`).

Based on the instruction, I set up the DIO8xLogicLevel below 
[image]

When I tried to start the controller from the codesys, I got the IO device responding but with `Wrong outputdata length: 0` message keeps showing.

```
pi@raspiprofinetB:~/profinet/build $ sudo ./pn_dev -vvvvv

** Starting P-Net sample application 0.2.0+v0.2.0-12-g6cd7a56 **
Number of slots:      5 (incl slot for DAP module)
P-net log level:      4 (DEBUG=0, FATAL=4)
App log level:        0 (DEBUG=0, FATAL=4)
Max number of ports:  1
Network interfaces:   eth0
Button1 file:
Button2 file:
Default station name: rt-labs-dev
Management port:      eth0 B8:27:EB:83:D3:ED
Physical port [1]:    eth0 B8:27:EB:83:D3:ED
Hostname:             raspiprofinetB
IP address:           192.168.0.69
Netmask:              255.255.255.0
Gateway:              192.168.0.1
Storage directory:    /home/pi/profinet/build

Init P-Net stack and sample application
Profinet signal LED indication. New state: 0
Network script for eth0:  Set IP 192.168.0.69   Netmask 255.255.255.0   Gateway 192.168.0.1   Permanent: 1   Hostname: rt-labs-dev   Skip setting hostname: true
Start sample application main loop

Plug DAP module and its submodules
Module plug indication
  Pull old module.    API: 0 Slot:  0
  Plug module.        API: 0 Slot:  0 Module ID: 0x1 "DAP 1"
Submodule plug indication.
  Pull old submodule. API: 0 Slot:  0 Subslot: 1
  Plug submodule.     API: 0 Slot:  0 Module ID: 0x1
                      Subslot: 1 Submodule ID: 0x1 "DAP Identity 1"
                      Data Dir: NO_IO  In: 0 bytes  Out: 0 bytes
Submodule plug indication.
  Pull old submodule. API: 0 Slot:  0 Subslot: 32768
  Plug submodule.     API: 0 Slot:  0 Module ID: 0x1
                      Subslot: 32768 Submodule ID: 0x8000 "DAP Interface 1"
                      Data Dir: NO_IO  In: 0 bytes  Out: 0 bytes
Submodule plug indication.
  Pull old submodule. API: 0 Slot:  0 Subslot: 32769
  Plug submodule.     API: 0 Slot:  0 Module ID: 0x1
                      Subslot: 32769 Submodule ID: 0x8001 "DAP Port 1"
                      Data Dir: NO_IO  In: 0 bytes  Out: 0 bytes
Done plugging DAP

Waiting for PLC connect request

Module plug indication
  Pull old module.    API: 0 Slot:  1
  Plug module.        API: 0 Slot:  1 Module ID: 0x32 "DIO 8xLogicLevel"
Submodule plug indication.
  Pull old submodule. API: 0 Slot:  1 Subslot: 1
  Plug submodule.     API: 0 Slot:  1 Module ID: 0x32
                      Subslot: 1 Submodule ID: 0x132 "Digital Input/Output"
                      Data Dir: INPUT_OUTPUT  In: 1 bytes  Out: 1 bytes
PLC connect indication. AREP: 1
Event indication PNET_EVENT_STARTUP   AREP: 1
PLC write record indication.
  AREP: 1 API: 0 Slot:  1 Subslot: 1 Index: 123 Sequence:  2 Length: 4
  Writing parameter "Demo 1"
  Bytes: 00 00 00 01
PLC write record indication.
  AREP: 1 API: 0 Slot:  1 Subslot: 1 Index: 124 Sequence:  3 Length: 4
  Writing parameter "Demo 2"
  Bytes: 00 00 00 02
PLC dcontrol message (The PLC is done with parameter writing). AREP: 1  Command: PRM_END
Event indication PNET_EVENT_PRMEND   AREP: 1
  Set initial input data and IOPS for slot  0 subslot     1 IOXS_GOOD size   0 "DAP Identity 1"
  Set initial input data and IOPS for slot  0 subslot 32768 IOXS_GOOD size   0 "DAP Interface 1"
  Set initial input data and IOPS for slot  0 subslot 32769 IOXS_GOOD size   0 "DAP Port 1"
  Set initial input data and IOPS for slot  1 subslot     1 IOXS_GOOD size   1 "Digital Input/Output"
  Set initial output         IOCS for slot  1 subslot     1 IOXS_GOOD          "Digital Input/Output"
Application will signal that it is ready for data, for AREP 1.
Event indication PNET_EVENT_APPLRDY   AREP: 1
PLC reports Provider Status (IOPS) 118 for input slot 1 subslot 1 "Digital Input/Output".
  Is the PLC in STOP mode?
Wrong outputdata length: 0
Setting outputs to default values.
PLC ccontrol message confirmation (The PLC has received our Application Ready message). AREP: 1  Status codes: 0 0 0 0
Wrong outputdata length: 0
Setting outputs to default values.
Wrong outputdata length: 0
Setting outputs to default values.
Wrong outputdata length: 0
Setting outputs to default values.
Wrong outputdata length: 0
Setting outputs to default values.
Wrong outputdata length: 0
Setting outputs to default values.
Wrong outputdata length: 0
Setting outputs to default values.
Wrong outputdata length: 0
Setting outputs to default values.
Wrong outputdata length: 0
Setting outputs to default values.
Wrong outputdata length: 0
Setting outputs to default values.
```

I also checked the LED text, which is the output. The value is still 0 and did not change at all.

```
Every 0.1s: cat /home/pi/profinet/build/pnet_led_1.txt        raspiprofinetB: Thu Dec 22 06:22:26 2022

0

```

Can someone points me to where did I go wrong with the setup? Any help will be appreciated.

dddomodossola, 2023-03-14:
@ibndias we have the same problem. Did you got it working?

ibndias, 2023-03-15:
> @ibndias we have the same problem. Did you got it working?

Nope, decided to move on and use codesys to codesys instead.

dddomodossola, 2023-03-15:
We ( me and @nowire ) debugged the code today, the outputdata is populated correctly. It is the sample code that writes to debug file that should have a bug. But the data is exchanged correctly. Thank you for replying.

ibndias, 2023-03-15:
Great! Good to know that this is not an issue, I will close it then. ;)