rtlabs-com/p-net #554: Question - How to read/write multiple modules/submodules which have the same structure?

rruiter87, 2023-05-24:
I have one base module defined in a GSDML file that has a submodule with a single in and output. This submodule can be plugged into any of the four subslots. Lets imagine the single input is an axis position setpoint and the output is the actual axis position.

Now I want to add multiple submodules, because I have multiple axis. How can I achieve this? Is the only way to create multiple modules or submodules in a gsdml file, such that each one has a unique module id or submodule id? 

I thought it would maybe also be possible to add multiple profinet devices from the same GSDML file into the project:

[image]

or add the same submodule to different slots, to have multiple axis:

[image]

But if I look into the code, there seems no way to id where the data belongs to, other than the module/submodule ids.

pyhys, 2023-06-01:
Hi! 
One instance of p-net corresponds to one device in your top picture. I think of different devices as different boxes each with an Ethernet cable. So it is possible to make two boxes each running p-net, but the data will not end up in the same program.

I think you should model your axes as modules, and to plug these modules into slots in the TIA portal like the lower picture.
If you would the modules (axes) to be identical, define a single module in the GSDML file and state which slots this module type can be plugged into. If the axes are different, define different modules in the GSDML file.

In for example pnet_output_get_data_and_iops() you can read out the data from each slot, 
At startup you will have a callback pnet_exp_module_ind() describing which type of module is plugged into each slot.

rruiter87, 2023-06-01:
Hi!

Thanks for the clarification. With these instructions I should be able to make it work.

Roald

closed 2023-06-02 by pyhys