rtlabs-com/p-net #571: Question:  Gathering Data From Profinet PLC's

Oskee-Wow-Wow, 2023-12-28:
New to the world of Profinet and PLC's and I have been assigned to collect various pieces of information from several different machines which have PLC's that communicate using Profinet - and then aggregate that data for reporting.

So, I need to be able to connect to the various PLC's from the device side asynchronously and query certain registers for values.

While searching the internet for either a profinet controller or supervisor open-source code to do this, I've come across p-net several times and I'm wondering if p-net as device code will accomplish this task?   It'll be running from a Linux box using Rocky-8.

Also, if p-net can help me accomplish this task, what constraints do I face using p-net in terms of hardware that i can connect to and limitations in software functionality?

Thanks!

pyhys, 2024-02-21:
Hi!
You typically use OPC-UA to connect to PLCs to collect information. 

The p-net library is a Profinet device (slave) library, and can unfortunately not do this.

pyhys, 2024-04-17:
Closing due to inactivity. Please reopen if necessary.

closed 2024-04-17 by pyhys