rtlabs-com/p-net #586: CMakeLists.txt missing

XiaoP76, 2024-08-20:
The tutorial https://rt-labs.com/profinet/a-beginners-guide-to-profinet-get-started-in-60-minutes/ suggests that there is a CMakeLists.txt.

Yet, if I try to build p-net according to the tutorial, I get the error message:
<code>
d:\Git\profinet>cmake -B build -S p-net
CMake Error: The source directory "D:/Git/profinet/p-net" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
</code>
which is about right, because the file is not there.

saemideluxe, 2024-09-10:
According to the commit messages the `CMakeLists.txt` was removed on purpose.

@maintainers: Was this on purpose to make the project un-buildable for the public? If so, it would be great to make this clear somewhere on the readme.

@XiaoP76 I think you could try to checkout an earlier commit to build the project.

saemideluxe, 2024-09-10:
@XiaoP76 Maybe worth checking out https://naresh97.github.io/OpenProfinet/

mekerhult, 2024-09-10:
Thank you for reaching out and for your interest in the P-Net stack. We understand your frustration, and we’d like to clarify the recent changes.

As a small team, we’ve invested a lot of time and effort into developing and maintaining the P-Net stack over the years. To continue offering high-quality support and improvements, we’ve had to make the decision to transition to a model that allows us to dedicate the necessary resources. While this means some parts of the codebase are no longer public, it allows us to offer a more reliable and professional solution for those who need it.

Looking ahead, we’ll be updating the license in a future commit, transitioning to a source available model. However, please note that the current master branch will no longer be updated and may eventually be removed. We are also in the process of updating the documentation to reflect all these changes.

Feel free to reach out. We’re happy to explore how we can help you with support or licensing options to ensure you can continue your work. Please contact: supportATrt-labs.com

Thank you for your understanding, and we’re here to support you however we can.
/RT-Labs team

saemideluxe, 2024-09-11:
@mekerhult Thanks for the response and the clarifications.
I completely understand the burden of maintaining an open source project and I respect your decision.
Part of the frustration comes from the situation that PROFINET seems to have almost no publicly available device-stack implementations (unlike MODBUS, which has quite a few), and this repo seems to have been the main place for people looking for such an implementation.

marioles50, 2024-10-29:
Is it possible to debug the p-net stack with Netbeans?
If so, how can I do it?
thanks

closed 2025-06-10 by elupus